bambulab/BambuStudio #583: Model name boxes in 01.03.00.25 very irritating

randyre, 2022-11-08:
The new version of studio now shows the model name in a box in front of the model, and all the model parts in multi-part models have these boxes which completely cover the model and are impossible to work with.

Please provide a toggle on the right side, or a button inside preferences to turn this on and off.

randyre, 2022-11-12:
**Are we going to address this?** I don't know why the labels even were a feature, it wasn't in the previous versions and I've not found a ticket asking for them.  They are just all sorts of in the way. I can't be the only one that finds this unacceptable. Does BL even use its own slicer?  I can hardly use it, especially for small parts, as it completely covers the part.

KD7LRJ, 2022-11-12:
In Prusa Slicer its hotkey is "E". This doesn't work in BS, but I wish it would. The feature is nice (sometimes) when trying to identify parts on a build plate. I can see how it would be unusable with lots of parts and no way to turn it off. I wasn't able to find anything in the UI to turn it on/off in BS.

[image]

Merlwynd, 2022-11-12:
Agree this should have a toggle off. I would never use it as typically and familiar with all the parts I print.

randyre, 2022-11-13:
@KD7LRJ @Merlwynd
Whew, thanks, thought I was going crazy and was the only one that thought these got in the way.

YifanWuBambu, 2022-11-30:
Will be optimized as not auto-show the label when switching to by-object sequence in the next release.

YifanWuBambu, 2022-12-16:
This is included in v1.4.

closed 2022-12-16 by YifanWuBambu