bambulab/BambuStudio #584: "Normal"="No Support"?

theogott, 2022-11-08:
[image]
[image]

Both pictures show the same Model.
The only difference is the setting of Support to "Normal".
Is it Ok, that in this case no support at all is generated?
Maybe the translation is wrong and "Normal" should be "No Support" ?

YifanWuBambu, 2022-11-08:
For "normal" type, support will be generated only at the place where is painted in "Support Painting" tool by users.
[image]
[image]
[image]

theogott, 2022-11-08:
Thank you for the explanation, in this case i would suggest that the Spelling "Normal" and "Tree" is misleading.
Maybe you can add "User Normal" and "User Tree" to make clear that this is the setting for "User Generated Support".

Secondly could you take a look on these significant horizontal lines in the Slicing?
Where do they come from?
[image]
[image]
[image]

YifanWuBambu, 2022-11-09:
@theogott 
Thanks for the suggestion. Maybe we will change it to
1. Normal(auto)
2. Tree(auto)
3. Normal(manual)
4. Tree(manual)

For the slicing issue, could you share your model with us?

theogott, 2022-11-09:
Hallo Yifan Wu,
thanks for your fast and kind response.

I will like to share this nice chinese dragon model with you, but it has 138 MB so i can not attach it here, 
You can download it from WeTransfer for the next few days using this Link:
[WeTransfer-Link](https://wetransfer.com/downloads/7927c93ef837db85809144bdde56e0b120221109071815/0301895f5356d9fff5ff5f398e6968ea20221109071856/774ac2 )

Now let me share the real printout and the picture in the slicer.

[image]

[image]

YifanWuBambu, 2022-11-09:
Thanks. We will fix it in the next release.

YifanWuBambu, 2022-11-10:
We will check it.

theogott, 2022-11-10:
Thank you Yifan Wu, if you have written it down, you can close the case here.

YifanWuBambu, 2022-11-11:
@tangketan will close it after it has been fixed.

tangketan, 2022-11-14:
The option names have been changed.
Another risk in this project is the normal support used. Since normal support won't grow its horizontal span, too tall support is very risky. I would suggest you to use hybrid(auto) or tree(auto).
[image]

closed 2022-11-14 by tangketan

theogott, 2022-11-14:
Could you remove the Lines in the support?
If a support thickness of 1 line is to small for large supports, maybe you can print the lower part with multiple lines and the upper part with 1 Line?