bambulab/BambuStudio #585: Allow Model To "Sink" Into Build Plate Without Resetting To First Contact Point

MadMax389, 2022-11-08:
In Prusa Slicer, I can lower the model below the print surface and then slice it. I do this sometimes to make sure that there is a flat 1st layer. The workaround is to use the cut feature which works well on a single-color model. However, if you cut a painted muti-color model, the slicer will remove all of the painting requiring the user to repaint. This can be time consuming. This becomes especially useful when importing previously painted models from PS.

bolsoncerrado, 2022-11-08:
I do use that feature too in my "other" slicer (S3D) so this would be great to have indeed!

MadMax389, 2022-11-08:
I used to LOVE S3D. Alas, no updates since 2019.

RobinWhitfield, 2022-11-09:
So the alternate fix here is to enable cutting without losing painting?

MadMax389, 2022-11-09:
Yes, or cut the model before painting. In Prusa Slicer you can drag the model below the plate and slice/print it.

[image]
[image]

closed 2023-04-18 by SaltWei