bambulab/BambuStudio #589: [Feature Request] Local printing

ACJUK, 2022-11-08:
This might be possible already but I'm assuming my job goes from the slicer to the cloud and then to the printer.  I'd like to cut the cloud out and just send the job locally

RobinWhitfield, 2022-11-09:
Existing feature. You just need to enable LAN mode on the printer.

YifanWuBambu, 2022-11-11:
Hi @ACJUK,
Have you tried the LAN mode? Does it meet your requirement?

julie777, 2023-02-23:
This should stay open until the P1P supports LAN mode.

lanewei120, 2023-03-03:
P1P support lan mode firmware released yesterday

closed 2023-03-03 by lanewei120

julie777, 2023-03-07:
I spoke too soon. 
This might be a slightly different problem. I just switched to Bambu Studio to check on a print and notice that the camera wasn't updating so I click on the play icon. 
I switch back to the browser to continue typing for a minute and when I went back to Bambu Studio it was hung. Then I noticed that I had another instance of Bambu Studio running and it had the camera image displayed. But it was also hung. 

I am not sure how I started the second instance of Bambu Studio. It was not intentional. It does seem to imply that my new hang was caused by trying to access the camera from a second instance. 

The camera has been working fine for the last couple days which seems to imply that the old bug is fixed. 

BTW, I have been using LAN only mode for the last couple days without a problem.