bambulab/BambuStudio #590: [Feature Request] Queue Print by Plate

tamag0chii, 2022-11-08:
If we have multiple plates (or can clean up quickly) we can queue the prints so after one plate/set is done we can swap out a new plate and click resume.

YifanWuBambu, 2022-11-11:
@tamag0chii 
We will add this feature in the next release.

YifanWuBambu, 2022-12-16:
This is included in v1.4.

closed 2022-12-16 by YifanWuBambu

Ali229, 2023-08-25:
This isn't exactly implemented in 1.4, we can have plates yes but we can't queue the prints, example if I have 10 plates and I just finished printing plate 9, I have to click print again>scroll 9 times to the 10th plate, then print.

I think a more elegant solution would be what OP mentioned, the print finishes, pauses, or stops, has a print next button, and automatically prints the next plate in queue.