bambulab/BambuStudio #592: [Feature Request] Chamber Temp Alarm when Printing PLA

tamag0chii, 2022-11-08:
An enclosed printer is very useful because it minimizes dust accumulation in the fans. The dust accumulation in the fans degrade it faster and will result in heatcreep faster.

Bambulabs X1 is great because it is enclosed. But when printing with PLA using Textured PEI Plates the Bed Temp usually is at a higher temperature and it might increase chamber temp to more than 45Celsius which softens PLA.

Alarm when Chamber Temp either executes:
- alarm in handy app
- or; as an option pauses print until chamber temp goes down?

YifanWuBambu, 2022-11-11:
@tamag0chii 
We will add a warning if bed temp exceeds 55 C for PLA before sending a print job.

bani6809, 2022-11-13:
warning should be for chamber temp, not bed temp.

YifanWuBambu, 2022-12-16:
This is included in v1.4.

closed 2022-12-16 by YifanWuBambu