bambulab/BambuStudio #601: Strange VPN Issues

Benderrodrigu3z, 2022-11-09:
so the VPN issue has been HALF fixed

if the vpn is on with local lan traffic I can send a print, however in order to find/see the printer I have to hit:
Print then Refresh, then Send and I might have to hit Send a second time cause the first time usually fails. Not sure what would cause this though but its happened twice so far, only have sent 2 print jobs though so it looks to be a bug

VPN Service
Windscribe with Allow LAN traffic turned on

macOS Catalina 10.15.7 2012

Benderrodrigu3z, 2022-11-09:
[image]
Closing the print screen and saying print again however works, there is a delay for the upload but its only an additional 5-10 sec then it sends it

StoneLiBambulab, 2022-11-17:
@Benderrodrigu3z 
Could you help to upload your bambu studio log, thanks!
Please refs to https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

Thanks!
Stone Li

Benderrodrigu3z, 2022-11-17:
[debug_network_Fri_Nov_04_20_44_05.log](https://github.com/bambulab/BambuStudio/files/10027431/debug_network_Fri_Nov_04_20_44_05.log)
[debug_network_Fri_Nov_11_14_39_14.log](https://github.com/bambulab/BambuStudio/files/10027432/debug_network_Fri_Nov_11_14_39_14.log)
[debug_network_Fri_Nov_11_14_48_00.log](https://github.com/bambulab/BambuStudio/files/10027433/debug_network_Fri_Nov_11_14_48_00.log)
[debug_network_Fri_Nov_11_15_46_02.log](https://github.com/bambulab/BambuStudio/files/10027434/debug_network_Fri_Nov_11_15_46_02.log)
[debug_network_Mon_Nov_07_00_19_23.log](https://github.com/bambulab/BambuStudio/files/10027435/debug_network_Mon_Nov_07_00_19_23.log)
[debug_network_Mon_Nov_07_00_19_49.log](https://github.com/bambulab/BambuStudio/files/10027436/debug_network_Mon_Nov_07_00_19_49.log)
[debug_network_Mon_Nov_07_00_21_19.log](https://github.com/bambulab/BambuStudio/files/10027437/debug_network_Mon_Nov_07_00_21_19.log)
[debug_network_Mon_Nov_07_00_22_15.log](https://github.com/bambulab/BambuStudio/files/10027438/debug_network_Mon_Nov_07_00_22_15.log)
[debug_network_Mon_Nov_07_00_25_05.log](https://github.com/bambulab/BambuStudio/files/10027439/debug_network_Mon_Nov_07_00_25_05.log)
[debug_network_Mon_Nov_07_00_29_55.log](https://github.com/bambulab/BambuStudio/files/10027440/debug_network_Mon_Nov_07_00_29_55.log)
[debug_network_Mon_Nov_14_23_17_26.log](https://github.com/bambulab/BambuStudio/files/10027441/debug_network_Mon_Nov_14_23_17_26.log)
[debug_network_Sat_Nov_12_16_41_21.log](https://github.com/bambulab/BambuStudio/files/10027442/debug_network_Sat_Nov_12_16_41_21.log)
[debug_network_Sat_Nov_12_18_18_30.log](https://github.com/bambulab/BambuStudio/files/10027443/debug_network_Sat_Nov_12_18_18_30.log)
[debug_network_Sat_Nov_12_18_47_54.log](https://github.com/bambulab/BambuStudio/files/10027444/debug_network_Sat_Nov_12_18_47_54.log)
[debug_network_Sat_Nov_12_18_49_57.log](https://github.com/bambulab/BambuStudio/files/10027445/debug_network_Sat_Nov_12_18_49_57.log)
[debug_network_Sun_Nov_06_23_42_16.log](https://github.com/bambulab/BambuStudio/files/10027446/debug_network_Sun_Nov_06_23_42_16.log)
[debug_network_Sun_Nov_13_12_44_14.log](https://github.com/bambulab/BambuStudio/files/10027447/debug_network_Sun_Nov_13_12_44_14.log)
[debug_network_Sun_Nov_13_20_12_10.log](https://github.com/bambulab/BambuStudio/files/10027448/debug_network_Sun_Nov_13_20_12_10.log)
[debug_network_Thu_Nov_03_14_38_36.log](https://github.com/bambulab/BambuStudio/files/10027449/debug_network_Thu_Nov_03_14_38_36.log)
[debug_network_Thu_Nov_03_20_54_03.log](https://github.com/bambulab/BambuStudio/files/10027450/debug_network_Thu_Nov_03_20_54_03.log)
[debug_network_Tue_Nov_01_21_49_24.log](https://github.com/bambulab/BambuStudio/files/10027451/debug_network_Tue_Nov_01_21_49_24.log)
[debug_network_Tue_Nov_08_09_31_30.log](https://github.com/bambulab/BambuStudio/files/10027452/debug_network_Tue_Nov_08_09_31_30.log)
[debug_network_Tue_Nov_08_09_51_19.log](https://github.com/bambulab/BambuStudio/files/10027453/debug_network_Tue_Nov_08_09_51_19.log)
[debug_network_Tue_Nov_08_19_09_36.log](https://github.com/bambulab/BambuStudio/files/10027454/debug_network_Tue_Nov_08_19_09_36.log)
[debug_network_Wed_Nov_02_09_58_45.log](https://github.com/bambulab/BambuStudio/files/10027455/debug_network_Wed_Nov_02_09_58_45.log)
[debug_network_Wed_Nov_02_12_14_25.log](https://github.com/bambulab/BambuStudio/files/10027456/debug_network_Wed_Nov_02_12_14_25.log)
[debug_network_Wed_Nov_02_23_07_50.log](https://github.com/bambulab/BambuStudio/files/10027457/debug_network_Wed_Nov_02_23_07_50.log)
[debug_network_Wed_Nov_09_09_52_00.log](https://github.com/bambulab/BambuStudio/files/10027458/debug_network_Wed_Nov_09_09_52_00.log)
[debug_network_Wed_Nov_16_17_36_08.log](https://github.com/bambulab/BambuStudio/files/10027459/debug_network_Wed_Nov_16_17_36_08.log)
[debug_network_Wed_Nov_16_21_15_12.log](https://github.com/bambulab/BambuStudio/files/10027460/debug_network_Wed_Nov_16_21_15_12.log)

StoneLiBambulab, 2022-11-18:
Hi @Benderrodrigu3z ,
      there are several VPN software known blocked bambu studio.
refs to https://wiki.bambulab.com/en/software/bambu-studio/failed-to-get-network-plugin

when the vpn is enable, bambu studio can not resolve the request host.
Please disable the vpn when using bambu studio.

Thanks!

Stone Li

Benderrodrigu3z, 2022-11-18:
What is the IP address for the Bambu server so that I can unblock it?

Sincerely,
Shane F

StoneLiBambulab, 2022-11-19:
@Benderrodrigu3z 
The IP address is not a certain one. Because this is a distributed service, it may be assigned to either server.

Thanks!

Stone Li

Benderrodrigu3z, 2023-03-11:
This bug is back on 1.5, nothing sends until you turn off the VPN