bambulab/BambuStudio #603: [Feature Request] Allow zooming with the camera during print

TheCableGuy99, 2022-11-09:
Can I request a feature to allow us to zoom in with the mouse on the camera while printing? Not just zooming into the center but to be able to target specific areas?

Benderrodrigu3z, 2022-11-09:
technically if you take a screen shot you can afterwards, and the camera quality is actually good at zoomed in
+1 for zooming while printing though

YifanWuBambu, 2022-11-11:
@TheCableGuy99 
The camera is equipped with a fixed-focus lens, after zoomed in, the image quality may not be very good.

TheCableGuy99, 2022-11-11:
> @TheCableGuy99 The camera is equipped with a fixed-focus lens, after zoomed in, it can only do digital zoom. The image quality may not be very good.

Yes I understand but it's going to be good enough to zoom a little and inspect little bits for a better inspection remotely. I have CCTV cameras which are fixed focus and I understand the trade-off but it's only for quick inspections.

TheCableGuy99, 2022-11-11:
thanks, would you like me to close the ticket or keep it open?