bambulab/BambuStudio #604: [Feature Request] Show current layer number

TheCableGuy99, 2022-11-09:
Can you add the current layer number and total number of layers in addition to the percentage of the print completed and time remaining?

Benderrodrigu3z, 2022-11-09:
+1 for this

YifanWuBambu, 2022-11-11:
This is in our plan and will included in recent release.

TheCableGuy99, 2022-11-11:
please put it on the mobile app as well :) thanks!

TheCableGuy99, 2022-12-16:
This wasn't added as promised?

Benderrodrigu3z, 2022-12-16:
It could be a printer firmware updateSincerely,Shane FSent from my iPhone13

TheCableGuy99, 2022-12-16:
Well it certainly needs writing into the slicer, but yeah it could possibly need a firmware update for the API as well.

spuiuk, 2023-01-25:
I noticed that this feature has been added in v1.4.2. Point 3
https://github.com/bambulab/BambuStudio/releases/

However it mentions X1/X1C and will rely on a future firmware update. I use a P1P and would like to see this feature included there too.

timothyzalusky, 2023-02-07:
[https://wiki.bambulab.com/en/x1/manual/X1-X1C-AMS-firmware-release-history Added support for displaying the current layer number/total number of layers during printing](https://wiki.bambulab.com/en/x1/manual/X1-X1C-AMS-firmware-release-history)

Issue can be closed

closed 2023-03-01 by lanewei120