bambulab/BambuStudio #605: Linux Camera/Video Stream Support

norpol, 2022-11-09:
Creating separate ticket from #12 to track video/camera support.

At the moment clicking on "play" will show `Loading failed [2]`.

As far as I can tell so far is that there is [`MediaPlayCtrl.c`](https://github.com/bambulab/BambuStudio/blob/c24680e09dc26c6f734e448fa6ef1b236a3d48ba/src/slic3r/GUI/MediaPlayCtrl.cpp) and [`MediaFilePanel.cpp`](https://github.com/bambulab/BambuStudio/blob/c24680e09dc26c6f734e448fa6ef1b236a3d48ba/src/slic3r/GUI/MediaFilePanel.cpp).

Doing `strace -o /dev/stdout -e writev -f -s 1024 /usr/bin/bambu-studio-bin --debug 0 |& grep -iF -e 'camera_url'`

will show after hitting the play button for multiple times:

```
4180799 writev(11, [{iov_base="\n2022-11-09 17:51:50.626768[Thread 0x00007f1e1d334e80]:process_ui_task{ type:Backup, id:0, path:, object:0, extra:37}\n2022-11-09 17:51:56.811586[Thread 0x00007f1e1d334e80]:MediaPlayCtrl::SetStatus: Initializing...\n2022-11-09 17:51:57.566416[Thread 0x00007f1d38f2a6c0]:camera_ur : bambu:///XXXX?authkey=XXX&passwd=XXX&region=us\n2022-11-09 17:51:57.566778[Thread 0x00007f1e1d334e80]:MediaPlayCtrl::SetStatus: Loading...\n2022-11-09 17:51:57.567121[Thread 0x00007f1d7affd6c0]:MediaPlayCtrl: start load\n2022-11-09 17:51:57.567844[Thread 0x00007f1d7affd6c0]:MediaPlayCtrl: end load\n2022-11-09 17:51:57.595905[Thread 0x00007f1e1d334e80]:MediaPlayCtrl::onStateChanged: 0, last_state: 5\n2022-11-09 17:51:57.595933[Thread 0x00007f1e1d334e80]:MediaPlayCtrl::onStateChanged: size: 0x0\n2022-11-09 17:51:57.595950[Thread 0x00007f1e1d334e80]:MediaPlayCtrl::onStateChanged: 0, last_state: 5\n2022-11-09 17:51:57.595954[Thread 0x00007f1e1d334e80]:MediaPlayCtrl::onStateChanged: size: 0x0\n2022-11-09 17:51:57.596046[Thr"..., iov_len=8076}, {iov_base="2022-11-09 17:53:00.625589[Thread 0x00007f1e1d334e80]:process_ui_task{ type:Backup, id:0, path:, object:0, extra:44}", iov_len=116}], 2) = 8192
```

Seeing the code it shows something Port 6000.

https://github.com/bambulab/BambuStudio/blob/c24680e09dc26c6f734e448fa6ef1b236a3d48ba/src/slic3r/GUI/MediaPlayCtrl.cpp#L126

nmapping doesn't show anything on that port going on. I know there is some local tunnel setup, perhaps that's also used for the stream?

* How does bambu-studio initialize retrieving the video stream?
* Would it be possible to allow streaming the video without bambu-studio (via web-browser, gstreamer/ffmpeg/mpv/vlc)?

lanewei120, 2022-11-14:
currently the linux liveview is not ready
all these functions are missed
and it is in our feature plan

also the video stream is from internal tunnel,
we can not play it directly via web-browser

thanks

hisptoot, 2022-11-14:
I just download libBambuSource.so from their server. Setup up a docker container, call functions of libBambuSource.so to get raw video data and feed the data to ffmpeg to get a liveview access.

lanewei120, 2023-03-01:
already added in recent versions

closed 2023-03-01 by lanewei120