bambulab/BambuStudio #608: [Possible Bug] Menu opens on different monitor

TheCableGuy99, 2022-11-09:
Hi,

I have dual monitors and the default monitor is the right-hand monitor.

When I have BambuStudio open and maximised to that screen, then go to the top left "File" or drop-down arrow menu it loads on the top right of the left screen instead, instead it should load at the top left of the right screen when I clicked the "File" menu or drop-down arrow.

No other software I have installed behaves like this so am confident it's isolated to BambuStudio.

Thanks.

theogott, 2022-11-10:
Already reported, See:
https://github.com/bambulab/BambuStudio/issues/550

TheCableGuy99, 2022-11-10:
Apologies, I did search but it didn't appear on my search.

Benderrodrigu3z, 2022-11-10:
Do you mind closing this ticket @TheCableGuy99 ? its next to the word comment for you

TheCableGuy99, 2022-11-10:
Not all all, thanks.

closed 2022-11-10 by TheCableGuy99