bambulab/BambuStudio #611: [FEATURE] Sequential printing: Unable to edit/lock the sequence

H4kunaM4tata, 2022-11-10:
Take this as example, the sequence is wrong:

[image]

If I move things around, it just get worse.
Under normal circumstances, I would press E to see the sequence number as displayed here

[image]

Then move the files up and down, top one is 1, 2, 3 and so on.
At the moment that is not possible while the slicer somewhat messes the sequence around :(
We should be able to move the files up and down and disable the slicer from auto arranging them.

[image]

Thank you

H4kunaM4tata, 2022-11-20:
@YifanWuBambu this issue is actually making it impossible to slice small objects.

By default, the slicer should never display the object name like this.
This is a BambuSlicer behavior only. I cannot see the small objects because of this unnecessary naming.

[image]

I can see the sequence but I cannot edit the sequence so it is pointless.
[image]

PrusaSlicer keeps everything clean, as it should.
Cura works the same way.

[image]

If, only if I wanna print by objects and press E, the files name and sequence are displayed.
I can select the file name and move it around to change the sequence.
This basic feature doesn't work either on BS.

[image]

This is a bug, not a feature anymore.

tangketan, 2022-11-23:
Per-object printing is actually much more complicated than people usually think. We put a lot of effort to make sure collision won't happen during per-object printing.
1. Manually setting printing order is forbidden. Since the tool head is quite large and asymmetric (the side where micro lidar is installed is much larger than the other sides), manually setting the order without colliding the toolhead while keeping maximum build plate occupancy, is very difficult.
2. Cura or Prusa way of print ordering is not applicable for our printers. They assume the toolhead is symmetric, and they ignore many factors that may cause printing failure (e.g. the collision with rod, the different printing temperature of multi color objects), which is understandable since they don't have a core-xy printer yet.
3. I do agree the UI is not so friendly as it should be. We'll try to make it cleaner and easy to use.
4. @H4kunaM4tata : Back to your original question (the 1st picture), why do you think the sequence is wrong?

H4kunaM4tata, 2022-11-23:
@tangketan ,
For reference.

[image]

PrusaSlicer, the sequence is wrong. Do not cross printed objects.
By pressing E, it tells me the object sequence.
It should be 1, 2 and 3, not 1,3 and 2.

[image]

Easy fix, click and drag the object file.
Now we have 1, 2 and 3
The gantry won't travel over printed objects, no crashes.
It does not matter if the printer automatically detects the object height in theory, the users should be in control and not the slicer.

[image]

Bambu Slicer wanna start in the middle (1), then move to the back(2), then cross everything to finish at the front(3) (??????), like PrusaSlicer, fair.
It won't change the sequence because it believes that is correct.
I cannot drag the files to change the sequence like PrusaSlicer.
Not too mention that it display this pop-up all the time, it is a pain to move small objects because you canot see.
PrusaSlicer only display the pop-up when pressed E.

[image]

This print I had to keep moving things over and over until I gave up. The objects were not tall enough to crash with the gantry.
The pop-up issue, I cannot see the small clips. These pop-up are on all the time.

[image]

I need to spend a long time moving the objects around until the sequence is right, start at the front towards the back, do no cross printed objects. There is no click and drag, the slicer automatically sets a sequence and that is about it.
As you move the objects around, the slicer automatically messes with everything over and over again.

[image]

As I said in my previous comment, there is nothing complicated, just basic logic.

- Stop auto-processing the sequence, it is doing a terrible job
- Allows the user to click and drag the file which sets the sequence, that simple.

Thank you for looking into this.

H4kunaM4tata, 2022-11-24:
@tangketan Thank you for checking this out.
And thank you for being open minded about it.

You guys are doing a solid work, keep that up.

Thank you for the "show labels", it is driving me nuts.
By default it is disabled but if you set print by object, it gets enabled automatically and does not goes away if you set the print back to print by layer. You have to manually disable it.
I can disable it now, that is what matter haha

I guess now is the part where this ticket gets close and we wait for future releases?

Thanks a lot.

YifanWuBambu, 2022-12-16:
This is included in v1.4.

closed 2022-12-16 by YifanWuBambu