bambulab/BambuStudio #612: [Possible Bug] Loosing connection to printer

TheCableGuy99, 2022-11-10:
Hello,

When a print is started and monitoring it I keep getting an error "Failed to connect to the printer".

The connection starts off fine but after anything from 2 minutes to 10 minutes of monitoring I get this error and cannot see the telematics anymore. The webcam still works but everything else is not live / feeding me information.

I can re-select the printer and it does it again consistently after the same amount of time.

Attached a screenshot of when it's disconnected.
[image]

Thanks.

randyre, 2022-11-15:
This is happening constantly since about a week ago.  It's not just the slicer that loses connection, the printer drops off the WiFi randomly. The only way to fix is to reboot the printer, but then will randomly drop the connection again. sometimes within the hour sometimes not until the next day.

Also, a clue to this is that the camera feed is slow and lags way behind, but never used to a few weeks back, Something happened with how the Printer connects over WiFi in a recent update to either Slicer, firmware, or the cloud.

TheCableGuy99, 2022-11-15:
> This is happening constantly since about a week ago. It's not just the slicer that loses connection, the printer drops off the WiFi randomly. The only way to fix is to reboot the printer, but then [...]

With respect there is an issue with the printer losing the connection but your issue sounds more like you have a network inssue at your end. Check your WiFi channels against other local WiFi channels and check your Router or AP is fast enough. Some 2.4GHz are faster than others.

StoneLiBambulab, 2022-11-16:
Hi randyre，
    Could you help to upload your log of bambu studio. thanks!
Please refs to https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

Thanks!
Stone Li

randyre, 2022-11-16:
@StoneLiBambulab   Will do.

Also, as additional troubleshooting I told the printer to forget my network, and when I tried to set it back up it kept saying it could not connect after I input the password. I then rebooted the printer and attempted to get on the network again and it took the password just fine. I had to go to work, so have not had a chance to see if it remains connected.

randyre, 2022-11-17:
@StoneLiBambulab 
[log.zip](https://github.com/bambulab/BambuStudio/files/10026588/log.zip)

randyre, 2022-11-17:
@TheCableGuy99 
I guess I'm not understanding why you are so quick to dismiss my problem. So is this issue a network problem on your end also? I'm seeing the nearly same issue, but you say that my end is an issue with my network, but yours is a bug? Doesn't seem that you want the issue found if you are so quick to dismiss someone else having the same issue.

I understand that you're trying to help me, but you want a bug squashed, yet when I try and confirm that I'm seeing the same bug that you originally reported, it's now not a bug but something else on my end. I don't get it.

TheCableGuy99, 2022-11-17:
> @TheCableGuy99
>
> I guess I'm not understanding why you are so quick to dismiss my problem. So is this issue a network problem on your end also? I'm seeing the nearly same issue, but you say that my [...]

I only reported the software losing connection and re-selecting it brings it back. You described the printer losing connection and a laggy stream whilst connected as well. That's why I suggested it could be a network issue. You say it's not and I accepted that. No issues this end!

randyre, 2022-11-18:
Cool. thanks. I see what the issue was. I was having the exact same issue as you but also threw in the laggy issue, but should have made that a separate issue, or waited until the first bug went away before lumping them together.

StoneLiBambulab, 2022-11-30:
Hi @TheCableGuy99 ,

I will fix it in next release.
Thanks!

Stone Li

TheCableGuy99, 2022-12-16:
> Hi @TheCableGuy99 ,
> 
> I will fix it in next release. Thanks!
> 
> Stone Li

Just to update you on this, it still loses connection but it re-connects pretty quickly.

Not sure if you are happy with that or want to try to perfect it... that's your choice :)

YifanWuBambu, 2022-12-16:
This is included in v1.4.

TheCableGuy99, 2022-12-16:
> This is included in v1.4.

What is included? It still loses connection but reconnects.... is this okay with you?

matbroomfield, 2022-12-16:
Also having camera connection issues with the version released 16th december. Does not show printer screen at all during initialisation, then may show preview for short periods or may drop out for entire print.

matbroomfield, 2022-12-16:
Get a repeated cycle of messages "Loading...." "Stopped -20000"

julie777, 2023-03-17:
This is an ongoing problem that I have been trying to help diagnose in #1361.
The printer network communication is flaky. It frequently just disconnects from the network.
Looking at packets in Wireshark there are many errors.

bani6809, 2023-05-21:
I had lots of connect and disconnect and stability issues with my x1c. I suspected it might be related to my Ubiquiti UAP-AC-PRO. I replaced it with a TP-Link EAP610 and not only did all my x1c problems go away, I can now get strong reception all the way inside my garage through three walls. I feel like I should be wearing lead lined underwear now, but it solved all my wifi problems. Maybe something to consider.