bambulab/BambuStudio #614: Auto Turn fans on self cool nozzle

Benderrodrigu3z, 2022-11-11:
I set the nozzle to 270 to ooze out extra filament, then told it to home, it set the nozzle head to 140 but didn't turn on the cooling fan so I have to wait longer, seams like a waste of a fan, perhaps turn it on if above 150, at 150 turn it off, this can be used in more places then this, such as at the end of a print, even greater to move the nozzle head over to the fan spout.
Side fan can also be used to cool off the bed when its too hot at the start of a print...

Benderrodrigu3z, 2022-11-11:
Or even a move print head to purge location now raise bed turn on side fan to help it detach and not micro bubble the bottom of the cold plate

YifanWuBambu, 2022-11-11:
@Benderrodrigu3z 
This is a firmware feature. Please move  to https://forum.bambulab.com for better support.

Benderrodrigu3z, 2022-11-15:
too much work, I'll let the idea die here