bambulab/BambuStudio #616: Add One More Pre-Defined Print Speed.

wirlybird, 2022-11-11:
In the selection box for print speed silent is 50% of standard.

I'd like to see one that is 25% of standard.
This would be convenient for small parts that just need to go real slow.

YifanWuBambu, 2022-11-11:
@wirlybird 
If you have a model that fails in the printing, please share it with us. That would be very helpful to optimize the default settings.

wirlybird, 2022-11-11:
HI,

I wasn’t about a failed model but just having the ability to select a real slow speed for those times it may be good.

This would make it a little easier than changing a bunch of settings!

 

Thank you for considering this.

YifanWuBambu, 2022-11-12:
@wirlybird 
For this kind of low speed, does changing speed parameters okay for you?
It is free to lower the speed and acceleration settings in detail on the side bar.
[image]

wirlybird, 2022-11-12:
Hi,

Yes, that is fine and I make a profile for that.  I just though users might like an easy to use speed setting one step below the “silent” one.

It was just a suggestion but I think there are many more items that are more important!

 

Thank you for taking the time to consider it.

 

 <https://user-images.githubusercontent.com/110661856/201478406-8ab81b35-ae54-461b-9695-94292d29a4ba.png> image

YifanWuBambu, 2022-11-15:
@wirlybird 
We will consider about that in new release. Thanks for the suggestion: )

suit, 2023-03-02:
this might be also loosly connect to this feature:
https://github.com/bambulab/BambuStudio/issues/1380

a more easy/simple to control global speed reduction without the need of chaning a boatload of values unter "speed"