bambulab/BambuStudio #619: Feature request: add chamber temp to bambu handy app

Armoth1, 2022-11-11:


YifanWuBambu, 2022-11-15:
@Armoth1 
X1/X1C do not have a sensor that aims to detect the chamber temperature, so we cannot get a relative accurate value.

KD7LRJ, 2022-11-15:
This is the display of the chamber temp from Bambu Studio. I'm not sure why my browser thinks the value is editable though.

[image]

It is also available in the Handy app when I tap on the "Printer" icon. It is displayed in blue in my Android version.

YifanWuBambu, 2022-11-15:
This is the display of the chamber temp from Bambu Studio. 
>> This is an approx value. I will sync with the app team for this requirement.

Why my browser thinks the value is editable though
>> Actually, it should be readonly. I reproduced the issue and it will be fixed in the next release.

Armoth1, 2022-11-15:
So the chamber temperature value shown on the printer screen is fake?! How
can you show a value on the printer then if there is no sensor?

Armoth1, 2022-11-16:
> It is an esimated temperature combined by several temp sensor in the
> printer.
> As I mentioned above, it is not so accurate as the nozzle and bed
> temperature, but you can get a rugh value.

Ok thank you

SaltWei, 2023-01-09:
Since this is not slicer issue. Close.
Please turn to https://github.com/bambulab/Bambu-Handy to report app feature request.

closed 2023-01-09 by SaltWei