bambulab/BambuStudio #620: [Feature] Color paint: Height Range - Allow user to specify exact height range or exact layer

candre23, 2022-11-12:
The height range feature would be a lot more useful if the user could type in an exact range in mm, or an exact range in absolute layers.  Many existing models are designed and provided with instructions for specific layer changes, and supporting these types of prints would be ideal.  Especially when the existing color engine is very janky - which it is.

Specifically, I'm requesting that text boxes be added to the height range window where a user could type in for example "layers 3 through 4" or "height 10mm through 10.5mm".

YifanWuBambu, 2022-11-15:
@candre23 
It was added to our plan list. Thanks for your suggestion.

Majestic7979, 2023-06-16:
@YifanWuBambu Is this going to be included, roughly how long?

YifanWuBambu, 2023-06-16:
@zhimin-zeng-bambulab 
Please help handle this request.

YifanWuBambu, 2023-06-16:
Hi @Majestic7979 ,

We added height range first. It may be used as an alternative tool for specifying color for a specific height range for now.
[image]

[image]

Majestic7979, 2023-06-17:
Thank you for responding, and it's great that we have that but I really need something more visual in the 3D viewer rather than the settings table.

zhimin-zeng-bambulab, 2024-06-26:
[image]
The latest version(v01.09.03.50) already includes this feature.

closed 2024-06-26 by zhimin-zeng-bambulab