bambulab/BambuStudio #621: [Feature] Color painting text

candre23, 2022-11-12:
Add the ability to paint text onto a model.  Seems simple and obvious, doesn't it?

YifanWuBambu, 2022-11-15:
Currently, Bambu Studio supports add text and quickly specify its color by numeric key #1~9.
Maybe you could use this feature first.

I have added text painting (texturing the text to a model) to our plan list.

candre23, 2022-11-15:
Yes, I am aware that text models can be added with the text tool and merged with other models to create embossed text.  But I am referring to simply "painting" text on to a model by changing the filament used for the region covered by the text, without adding or subtracting material.  

The circle paint tool is able to change the color/material for a surface region without modifying its geometry.  I am proposing a text paint tool which would do the exact same thing - except instead of the brush being a circle, it would be text of the user's choosing.

SaltWei, 2023-10-08:
[image]
You can set the text to be modifier so that the text surface can be aligned with model surface absolutely.
[image]

SaltWei, 2023-10-08:
> [image] You can set the text to be modifier so that the text surface can be aligned with model surface absolutely, just like painting [image]