bambulab/BambuStudio #623: [BUG]  New bambuStudio seems to break saving presets to cloud

psiberfunk, 2022-11-13:
In older versions of BambuStudio there seems to have been no issue saving and keeping in sync filament profiles.  Suddenly with [Bambu Studio v1.3.0.25](https://github.com/bambulab/BambuStudio/releases/tag/v1.3.0.25) , it seems that newly added user filament profiles don't save to the cloud.. only locally.  I tried making several new dummy ones and they don't update, to my great confusion.  I'm not sure what broke here, but the current version of bambu slicer is writing filament JSON files with this tag in them:

"version": "1.3.0.13"

Which seems to be out of date with the version of BambuSlicer itself.  Oddly, old filament profiles I have that have numbers like:

"version": "1.3.0.2"  or "version": "1.2.0.3"  seem to be importing from the cloud fine.  Only new ones seemingly won't sync.

psiberfunk, 2022-11-13:
P.S.  I tested extensively with two computers with the same up-to-date version of Bambu Slicer (1.3.0.25) on both, and tried lots of different orders of opening/quitting the application, as well as logging in and out, and checking and unchecking the "sync presets" in the preferences... nothing seems to get syncing to work for the newly created filament profiles.

psiberfunk, 2022-11-13:
An update:  I _think_ what may be causing the confusion here is this:  it seems like synchronization of deletes is weird somehow.   Imagine two computers, A and B , and a preset 'TEST'.

When you create 'TEST' on A it seems to sync to the cloud and on restarting BambuStudio on computer B, it appears.  Now, if I delete 'TEST' on B, it remains on A and does NOT get deleted (which it should be).  This means that if I then SAVE

 When you delete a preset from one computer, it remains on the second.. and then, even if you go back and update 'TEST' on A (say, change the temperature and save), it will STILL not show up in the cloud/get sync'd.

So the bug here is that user profile deletes arn't getting propagated I think, which causes a lot of confusion when you're trying to figure out what's going on.  As soon as those profiles are 'orphaned' in the cloud (deleted) the OTHER bambustudio instances should be cleaning them up at next launch, but are not.  This leads to a messy situation where you can have two profiles with the same name on different computers but one is cloud based, and the other is local only.

lanewei120, 2022-11-18:
hello

thanks for reporting this issue, we will refine this logic in next release