bambulab/BambuStudio #624: Feature Request

Anthorin, 2022-11-13:
Amend Studio to show the current layer being printed. In the event of a failure it would enable the model to be started at the failing layer.

Anthorin, 2022-11-13:
If a failure occurs, knowing the layer would mean being able to simply print the remaining part.

Benderrodrigu3z, 2022-11-15:
with a note to say I think I failed at this layer or a scan to scan to see where failure might had first occurred to finish the print

YifanWuBambu, 2022-11-15:
This is in our plan list. Thanks for your suggestion.

theogott, 2022-11-15:
Ok, this sounds really helpful. I had multiple cases where larger statues fail at the head.
Maybe this can be extended in Future to be able to "Split" large Objects in Studio into multiple Parts and print these an separate plates?
Most features needed for that are already builtin.
Lets say i want print a statue of 0,5 m in size. 
Studio could split the Object and spread it on 4 plates for printing.
This is nearly the same process as if you "continue print something - because you need to make a base-plate anyway.
Note that with Lightning Infill there is possibly only the hull otherwise depending on where the break occured.

YifanWuBambu, 2022-11-23:
> 

Yes. Will be like this.

closed 2023-03-01 by lanewei120