bambulab/BambuStudio #625: [BUG] BambuStudio has broken cloud sync: Certain new presets will not sync

psiberfunk, 2022-11-13:
During the course of investigating https://github.com/bambulab/BambuStudio/issues/623 , I came across a problem with creating a filament profile based on the Generic PA preset.  Such a new preset refuses to cloud-sync for some reason:

Replication steps:

1. Select the Generic PA preset.  
2. Edit the Preset, Make **no changes to the preset but save it as a new name** (simplest possible case).  In this case, I named mine 'TESTPA'  This generates the expected file in C:\Users\USER\AppData\Roaming\BambuStudio\user\USERID\filament  (see attached files)
3. The file refuses to cloud-sync.  It seems to have a blank "setting_id = " entry in the .info file, which is different from all the ones that cloud sync.

[filament.zip](https://github.com/bambulab/BambuStudio/files/9997798/filament.zip)

P.S.  This seems to be a problem for more than just the PA case.. profiles based on Generic PA-CF also do not work for some reason, and possibly other types as well.  It's not clear why?

StoneLiBambulab, 2022-11-21:
Hi @psiberfunk 
It has beed fixed. Please try again.

Thanks!

Stone Li

psiberfunk, 2022-11-21:
Acknowledged that this is fixed now .

YifanWuBambu, 2022-12-16:
This is included in v1.4.

closed 2022-12-16 by YifanWuBambu