bambulab/BambuStudio #628: More Filaments More Colors

Benderrodrigu3z, 2022-11-14:
Can we get more Color options? Personally I Need Rainbow, someone here with 4 AMS's hard to recall which one is Rainbow
Perhaps a this filament is multi colored, what's the first and second color and put both colors into one box so we know its the multi one

Can we get more brands added?
Such as Hatchbox, Sunlu, Gizmo, Tecbears

Benderrodrigu3z, 2022-11-14:
my concern is that the sync keeps the colors and destroys everything to Generic
we need a push method not just a pull, 4 is rainbow but how would I know unless I name it that.... and icon to show this please
[image]

YifanWuBambu, 2022-11-15:
This is in our plan and will be included in recent release.

theogott, 2022-11-15:
I remember that I ordered at the Bambu Store and the stuff was packaged in a Box from SUNLU, so i guess there is already a connection and maybe some of the Bambu Filaments are from SUNLU. So it should be added ...

ImagineerNL, 2022-11-17:
@YifanWuBambu I was writing a new issue, but this one touches upon my wishes, but not quite.
If this only fixes colors, this doesnt resolve other issues like multiple brands for same material, and even unlisted materials.

I have mainly eSun ABS+ and HIPS, which in AMS have to be labeled 'generic ABS' since eSun isnt listed as a brand, and HIPS isnt listed as a type, but its close to ABS.

It would be great if you could  select to 'sync this preset to AMS', for example by a button next to the filament, pointed in this picture
[image]

This is then a perfect route to upload and update extra profiles in the AMS.
First you tune it in the bambustudio with testprints, and then when it is finalized you can chose to upload it to the AMS to be used in filament-profile selection on both printer and bambustudio.

This also helps not needing to label HIPS as 'generic ABS' and give it its real name.

(Unless i'm missing another place where you can tune and update the specific profiles and colors in AMS.)

Benderrodrigu3z, 2022-11-17:
AMS doesn’t handle the colors or the filament type, that’s a firmware machine thing, remove the AMS and it’s still called generic …

Sincerely,
Shane F