bambulab/BambuStudio #629: Disable all lights when not in use?

FreeManFreeThinker, 2022-11-14:
When the device goes to sleep, it would be nice if the AMS lights and Bambu Lab logo on the hotend turned off. Or if there were a setting that'd be great. 

Or is it recommended to just turn off the printer when not in use

YifanWuBambu, 2022-11-15:
This is a firmware improvement. Please move to https://forum.bambulab.com/ for better support. Thanks!

candre23, 2022-11-15:
> This is a firmware improvement. Please move to https://forum.bambulab.com/ for better support. Thanks!

Full sleep functionality requires firmware modification, but you could at least add the option to turn out the lights after a print is complete and have the slicer add the appropriate commands to the end gcode.

AbeFM, 2022-11-16:
Agreed here. A firmware "sleep all lights when untouched for XX minutes" would be nice, but certainly after a print I could stand to have stuff shut off.