bambulab/BambuStudio #630: Odd hole bridging bug

ptyork, 2022-11-14:
A common technique to allow supportless printing of a recessed screw hole is to add a single layer, solid layer at the transition between head (or bolt) and the smaller bore for the screw itself. Generally, a slicer will add a solid base layer here that easily bridges the hole for the head/bolt. Then the screw hole is continued normally above this base layer, now fully supported. Once printed, this single layer is easily punctured for a very clean print.

Bambu handles this incorrectly. Instead of printing a base layer, it appears to perhaps start the base of the hole above it. The result is that it prints a filled circle completely unsupported.

See the comparison images. Left to right are layer before the transition, the base layer, and the subsequent start of the screw bore hole. Top to bottom are Cura, Bambu, and Prusa. Only Bambu fails, leaving an unpredicably-attached blob of plastic extruded below a now poorly formed screw hole.

[image]

SaltWei, 2022-11-15:
Thanks for reporting issue to us to keep improvement!

jeanblanchard, 2022-11-15:
I was about to report the same bug. Sacrificial bridges are a pretty common modeling trick.

Attached is a reduced test-case model that I just hacked together (CC-0). It shows the issue in BambuStudio, and works fine in PrusaSlicer (I uploaded a zip because GitHub does not like 3MF files. zip).

[Sacrificial Bridge v1.zip](https://github.com/bambulab/BambuStudio/files/10015882/Sacrificial.Bridge.v1.zip)

"Bridge" in BambuStudio: 
[image]

Bridge in PrusaSlicer
[image]

SaltWei, 2022-11-16:
Thank you very much for sharing!
This is a known issue which is caused by the option "Only one wall on top surface".
The fix will be included in next release.

YifanWuBambu, 2022-12-16:
This is included in v1.4.

closed 2022-12-16 by YifanWuBambu