bambulab/BambuStudio #631: Supports - even Tree-Support - break easily when printing large Statues

theogott, 2022-11-14:
You can clearly hear that _the nozzle is hitting the filament_ from time to time.
Would it help to calculate the thickness of supports structure depending on height?

When printing higher Objects this problem often occurs.
The only alternative would be to print without supports, but for this we would need a slightly longer nozzle (see video) as it needs 45 Degree air to all sides (but should be possible).

"**Printing without supports" using** **Conical Slicing**.
https://www.youtube.com/watch?v=1i-1TEdByZY

Maybe a partial adaption would also help?
The current results are not as it should be.

You can download the slice used below here (WeTransfer Link valid for few days): 
https://we.tl/t-0ZTXwtCqHW 

[image]
[image]

[image]

[image]
[image]

Benderrodrigu3z, 2022-11-14:
just here to help, what is the Tree Support Branch Distance and Diameter set to? if its breaking like this you may need to increase it

theogott, 2022-11-14:
I have attached the files, i think it was all "Standard". And i think "Standard" should work. Also I think the head should not hit the Filament so hard :-). Maybe the Software could make automatic settings depending on the height of the Object?
Because a normal customer may not know "When to change Support Branch Distance and Diameter".
But possibly it would help.

Benderrodrigu3z, 2022-11-15:
I find it personally better to set them to 1 if I want them to break off easily, the default is 5, the issue with height changing it is every print is different but it could guess if its not told what to do, ps you can save custom settings so you don't have to tell it from print to print

theogott, 2022-11-15:
Ok, default they are on 5 and now i have set them to 10 to test if this will hold according to your suggestion.
So you recommend to set them to 1, both?
Is there somewhere a detailed description of all these Parameters available?

theogott, 2022-11-15:
I have now changed  Support Branch Distance and Diameter both from 5 to 10.
The result is still not satisfying.

See ...
[image]
[image]
[image]
[image]
[image]

Benderrodrigu3z, 2022-11-15:
okay so you can try turning up tree support wall loops and maybe turning on Tree Support with infill, my tree support branch angle is set to 30, to make this a less filament test I recommend cutting the item in half vertically to do testing, this print looks like it has taken a long time

theogott, 2022-11-15:
Yes, this printing takes more then a day and its still printing. 
I will now set the speed to 50% maybe then it will do the rest ...
So which Parameters do you suggest for such large statues?

theogott, 2022-11-15:
I am from Germany, here this machine is much too expensive, look:
https://amzn.to/3X5DQxV

Maybe this one would work?
So you believe that i have a Dehydration problem with the filament?

Benderrodrigu3z, 2022-11-15:
https://www.amazon.com/gp/product/B07GR65CNT/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
this will tell you how much moisture is in the room or area, to print good you want less then 20%

Benderrodrigu3z, 2022-11-15:
come join the discord and I can send you the stl files in a dm
https://discord.gg/bambulab
my username in there is TheTexMech x3AMS

theogott, 2022-11-15:
You can find me as Theo_Gottwald
However it says "I must claim the Account". But it does not accept my password. 
Maybe this is too difficult for today. Because i can not reboot and the Discord itself seems not to start currently.
Maybe the Computer is too loaded.

Benderrodrigu3z, 2022-11-15:
I can't find or message you until your in the discord

Benderrodrigu3z, 2022-11-15:
once you get in and message me please hit the close issue button at the bottom

Benderrodrigu3z, 2022-11-22:
> You can find me as Theo_Gottwald
> However it says "I must claim the Account". But it does not accept my password.
> Maybe this is too difficult for today. Because i can not reboot and the Discord itself seems not to start currently.
> Maybe the Computer is too loaded.

did you ever join our group on the Discord?
https://discord.gg/bambulab
my username in there is TheTexMech x3AMS

tangketan, 2022-12-12:
@theogott  We have improved tree support in the next release. This problem should be fixed. However I didn't find the model to verify. Would be grateful if you could provice the model. Thanks~

theogott, 2022-12-12:
> Would be grateful if you could provice the model. Thanks~
The Upload-Link is via WeTransfer therefore the Link will expire after some days. So download it soon.
Try to print it out in a large sizen. If that works all is fine.

Let me add that possibly "Arc Overhangs" would make most or all Support obsolete.
So maybe you can add these for next release?
If that works this would give the X1 a Leap ahead.

See:
[Arc-Overhangs](https://github.com/bambulab/BambuStudio/issues/804)

**[Download the Model here]**(https://theogottwald.wetransfer.com/downloads/fc3e7d94c9090567793785350a50f50020221212094017/06f4ac192fc05d73ce0cf7c6cf9f075420221212094017/c74e2e )

tangketan, 2022-12-13:
Thanks! I've downloaded the file. There are a few reasons for the low strength:
1. object layer height is 0.12 and there are a lot of small height layers in supports.
2. the filament seems to be silk PLA? Silk PLA is known for its weak strength.
[image]

Here are some suggestions for now:
1. increase object layer height to 0.16 or 0.2mm
2. increase tree support wall count to 2
3. use some other more strong filament if possible

By the way, we've improved the tree support quite a lot (stronger support with less material), but we need more time to test it before releasing. I'll try the new algorithm and post some pictures later on (this model takes nearly a whole day to print).

theogott, 2022-12-13:
Just for my better Understanding.
Why do we invest time in more refined Support-Structures, if we can invest time in making them
completely Obsolete?

See "Arc Overhangs"
https://youtu.be/B0yo-o47688

tangketan, 2022-12-13:
Arc Overhang is an interesting attempt. It might be useful in some cases. But for now it's more experimental, as you can see the overhang surfaces are not quite quality. And I don't believe supports can be completely abandoned. Consider a unicorn like this. There is no way to print the tail with any of "Arc Overhang" or "Conical Printing". Even with this dragon model, supports are necessary.
[image]

theogott, 2022-12-13:
Yes, you are right. These U - Formed shapes will need Supports anyway.

tangketan, 2022-12-14:
Printing finished. Not bad :D
[image]
[image]

theogott, 2022-12-14:
Here is mine. In Multicolor it will be even better.
Why don't we use Multicolor?
Its still wasting too much Filament? :-)
[image]

theogott, 2022-12-17:
Hallo Tan

Just want to tell you that since the last release the 

Camera does not more send the video to my Windows PC.

 

So if you want see this Effect remote or check something remote,

or have a Beta to test i fit works, you can contact me here.

 

In such case we can also use Teamviewer or something like that.

I have checked that ist not a issue with the Windows Firewall.

 

 

 

 

Mit besten Grüßen |With best regards |Cordiali saluti

 

Theo Gottwald

Vitamin-Beratung * Betriebliche Gesundheitsvorsorge 

Smart-Package Robot

 

Tel. 07247-9851112  
mob.: 0160-6688222

Fax: 07247-9851113

 

Theo Gottwald

Herrenstr.11

76706 Dettenheim 

Germany (BRiD)

 

***@***.***

tangketan, 2023-02-09:
Did you mean the video stream to Studio on PC was not working? Do you still have the issue with the latest version now?

theogott, 2023-02-09:
This posting was from last december so the issue with the video Stream is solved.
There is a minor issues which is maybe personal, the stream will stop if i switch to another desktop and have to restart it when switching back to this desktop.
See: [Switching Desktops](https://www.youtube.com/watch?v=lxfHwN8mblc)

tangketan, 2023-02-10:
I'll transfer the desktop issue to another colleague.
I understand your curiosity to Conical Supports and we are keeping an eye on it, including other curved slicing techniques.

theogott, 2023-02-10:
Thank you Tang,

 

the Support is really the biggest Issue in 3D Printing, 

the more we can get rid o fit, the better stuff we can print. 

 

I have several things that can be printed, but are useless after removing the Supports.

Any solution will be a revolution in the market.

SaltWei, 2023-03-02:
v1.5.0 public beta has been released. Please let us known if still exist.
Thanks!

closed 2023-03-02 by SaltWei