bambulab/BambuStudio #633: Reduce Seam visibillity and better alignment

Itebayo, 2022-11-14:
One issue I have with the slicer are the very visible and sometimes not properly aligned seams. It would be awesome if you could get them to the visibility level that you get from Cura sliced objects.

As a little printfarm owner of decorative items I would love to see this improve.

AbeFM, 2022-11-16:
You mean when you check the "seam" box in the preview, you feel the white dots are too small?

Itebayo, 2022-11-16:
no that is fine. I mean on the printed out product, the seam is a lot more visible. 3d prints that were sliced with other slicers are less visible, especally on round objects. Looking at the percision of the printer they should look way better.

Try printing out two bowl-shapes. One sliced with bambu studio and one with Cura. The difference will be easy to see.

AbeFM, 2022-11-17:
Oh yeah, I totally agree. The seams stand out on Bambu prints more than other printers. Perhaps a "wipe" move, or just an end-of-loop overlap like infill gets.

QingZhangBambu, 2023-02-06:
speed of Bambu's printer is much faster than others. So the filament may accumulate in the seam. We will try to improve it by adjusting travel paths or others.

Itebayo, 2023-02-06:
I understand what you mean, but even if you set the speed and acceleration very low (e.g. Ender 3 level) the seams look way worse than they should. 
I have already posted some settings that seem to help in issue #1247.

Furthermore the machine should be able to produce even better seams if the alignment on round objects is actually straight, this is an issue that has been there for several patches.

You guys are still doing great work! I can't wait what will come out next, I am rooting for you!

QingZhangBambu, 2023-08-08:
@Itebayo we've added "Seam gap" and "wipe speed" parameters in out latest version. you can edit these parameters in develop mode. We hope that would be helpful.
[image]

closed 2023-10-16 by QingZhangBambu