bambulab/BambuStudio #637: enhancement: Allow Fuzzy Skin to be printed on a face

ipodsync, 2022-11-15:
In addition to being able to apply fuzzy skin to outside or all faces it would be nice to be able to specify the faces to apply it to.  

I'd love to leverage this ability, but the present options make impractical for what I'm printing.

Benderrodrigu3z, 2022-11-15:
there is a paint bucket tool, is that what your looking for?

JstBecause, 2022-11-16:
> there is a paint bucket tool, is that what your looking for?

A paint bucket tool, or brush, etc to choose which surfaces you want the fuzzy skin on would be a great addition.  I have a print I want to try just a few walls to be fuzzy, while keeping the inside (due to how tightly it fits around another piece) not fuzzy.

psiberfunk, 2022-11-17:
It feels like you can almost do it with the present options, but the modifier objects do stupid things like make extra lines.  Here's an example with a cylinder and a modifier cube, which makes only part of it fuzzy.. but introduces a new obnoxious set of walls in the middle of the cylinder.. to be fair, this is also a problem in upstream prusaSlicer.

[image]

Results in:

[image]

psiberfunk, 2022-11-17:
Interestingly, in prusaSlicer, you can do this with with a good effect with height range modifiers (but probably not what you want here).. which BambuSlicer sorely lacks.. I wish they would have it for other reasons:

[image]

JstBecause, 2022-11-17:
> It feels like you can almost do it with the present options, but the modifier objects do stupid things like make extra lines.  Here's an example with a cylinder and a modifier cube, which makes only [...]

Yeah this is a known bug in all these products. Or maybe not be a bug, but not how it is designed. Would work absolutely great if it did work the way it is supposed to! Would make a lot of peoples lives easier. 🤣

Benderrodrigu3z, 2022-11-17:
if you want to all be fuzzy then put it over the entire object? not sure why that's difficult.... again I don't work for Bambu

psiberfunk, 2022-11-17:
> if you want to all be fuzzy then put it over the entire object? not sure why that's difficult.... again I don't work for Bambu

That’s not what he’s asking for .  He’s asking for the ability to “paint” the fuzzy ness , which is a good idea .

SillacSaurfang, 2022-11-28:
I would absolutely love this, I've always wanted it from a slicer. Fuzzy skin as it works now is great for single-piece items, but manually blocking out the internals or interfacing parts of a complex or curved object with modifiers can be a nightmare.

Photogad, 2022-12-08:
Yes, PLEASE!

hutje, 2024-07-25:
Apparently this problem is still not fixed😔
Is there any possibility it would be released anytime soon?

magnusmyklebost, 2024-11-27:
Prusa have done it. Why cant bambu? 
https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1

hutje, 2024-11-27:
> Prusa have done it. Why cant bambu? https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1

Because Prusa has just released it in their alpha version. Give it time. Most new features are getting to all open source slicers eventually. Most likely OrcaSlicer will implement it first, then BambuStudio...

hutje, 2024-11-27:
I also hope it will eventually be possible to have the option to load a depth map image with a texture to emboss this in the surface. Now this process has a lot of steps by using Blender or specific 3D-cad software to emboss the object.
But this way you can create leather-like textures or woodgrains on an object.
[image]

magnusmyklebost, 2024-11-27:
Very cool, but i think fuzzy skin feature or other grip textures will be more usefull as this will help functional prints with gripping friction as well as being used for cosmetic reasons.

alie2n, 2026-01-15:
I just stumbled accross this issue and would also second this thought. Adding arbitrary fuzzy skin would be great.