bambulab/BambuStudio #638: Start Gcode issues

goldjunge91, 2022-11-15:
Hi
when create a Gcode there couple of lines that i don't want is there any option to remove it ?

; EXECUTABLE_BLOCK_START 
this i want to remove:
```
M73 P0 R123
M204 S500 ; adjust acceleration
```
this is my start code:
```
;TYPE:Custom
START_PRINT BED_TEMP=60 EXTRUDER_TEMP=220
```
i want to remove this all following befor G1 F900
```
M190 S60 ; set bed temperature and wait for it to be reached
M104 S220 ; set nozzle temperature
M109 S220 ; set nozzle temperature and wait for it to be reached
G90
G21
M83 ; only support relative e
; filament start gcode
M106 S0 ; disable fan
;LAYER_CHANGE
;Z:0.2
;HEIGHT:0.2
G1 E-5 F3600 ; retract
SET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY=8 ; adjust jerk
G1 X71.788 Y71.788 Z.2 F9000 ; move to first skirt point travel_to_xyz
G1 E5 F2400 ;  ; unretract
;TYPE:Skirt
;WIDTH:0.5
```
i want to remove this all 
G1 F900

I know that all ";" comments that don't need to remove only the set bed etc.
[export bambu.zip](https://github.com/bambulab/BambuStudio/files/10015199/export.bambu.zip)
Gcode and Profiles are all in the zip file

SaltWei, 2022-11-17:
So do you want to add profile for anycubic printer?

goldjunge91, 2022-11-17:
No, I already have this done. I would like to edit or remove the start gcode of bambustudio.
All of this lines 

> M190 S60 ; set bed temperature and wait for it to be reached
M104 S220 ; set nozzle temperature
M109 S220 ; set nozzle temperature and wait for it to be reached
G90
G21
M83 ; only support relative e
; filament start gcode
M106 S0 ; disable fan
;LAYER_CHANGE
;Z:0.2
;HEIGHT:0.2
G1 E-5 F3600 ; retract
SET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY=8 ; adjust jerk
G1 X71.788 Y71.788 Z.2 F9000 ; move to first skirt point travel_to_xyz
G1 E5 F2400 ;  ; unretract
;TYPE:Skirt
;WIDTH:0.5

SaltWei, 2022-11-17:
>M190 S60 ; set bed temperature and wait for it to be reached
>M104 S220 ; set nozzle temperature
>M109 S220 ; set nozzle temperature and wait for it to be reached
   These three line is generated to set temperature automatically. If you don't want it, you can add "M190 S0\n M104 S0\n" at the beginning of start gcode, and then studio will not generate these command. But that means you need to set temperature by other Macro.

The following part is generated by slicing rather than start gcode part, should has no influence on printing no matter what printer is used.
>G90
>G21
>M83 ; only support relative e
     Can't be removed unless modifying code. But this is necessary for bambu machine. If has no side effect, please don't change it.

>; filament start gcode
    Clean the custom gcode in filament setting page and this line will disappear
>M106 S0 ; disable fan
>;LAYER_CHANGE
>;Z:0.2
>;HEIGHT:0.2
       Can't be removed unless modifying code. But this is necessary for Gcode viewer. If has no side effect, please don't change it.

>G1 E-5 F3600 ; retract

>SET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY=8 ; adjust jerk
    This is not generated by Bambu Studio

>G1 X71.788 Y71.788 Z.2 F9000 ; move to first skirt point travel_to_xyz
>G1 E5 F2400 ; ; unretract
>;TYPE:Skirt
>;WIDTH:0.5