bambulab/BambuStudio #639: UI scaling is inconsistent

dotysan, 2022-11-15:
I have 2 displays in Windows 10.

Primary is 3840 x 2160 (Recommended) scaled to 200% (from recommended 250%).

Second is 3840 x 2160 (Recommended) scaled to 125% (from recommended 150%).

When in primary display it looks normal.
[image]

But not in second display.
[image]

Notice how the section name/headers are chopped and misaligned.

dotysan, 2022-11-15:
This is easily reproducible by moving the Bambu Labs window between displays.

SaltWei, 2023-03-02:
v1.5.0 public beta has been released. Please let us known if still exist.
Thanks!

closed 2023-03-02 by SaltWei