bambulab/BambuStudio #642: Bottom Z distance

JstBecause, 2022-11-16:
We have the option for top z distance, but not bottom z distance.

Some materials need more of a bottom z distance for easy removal. However we don't have this option.

For instance. TPU, when I print TPU and use supports, the Z distance needs to be between .6 to .8 for the lines to come off, and come off without scarring the print. If I printed a bottom z distance of .2 the support lines would **never** come off.

YifanWuBambu, 2022-11-17:
Hi @JstBecause ,
Thanks for your suggestion. We will add this in the next release.

MusangKingBambulab, 2022-11-28:
hi @JstBecause, thanks for your suggestion; we have added the option "Bottom Z distance"  for both normal support and tree support and will come out with the next release. Look forward to your feedback and other valuable suggestions.

YifanWuBambu, 2022-12-16:
This is included in v1.4.

closed 2022-12-16 by YifanWuBambu