bambulab/BambuStudio #644: [Feature] Optionally show the Calibration lines on the buildplate (different buildplate graphic)

ImagineerNL, 2022-11-16:
It would be a great help to have the calibration lines on the buildplate.

best route for it would be to include this option it in the buildplate image on the Printer Settings
("Printer settings tab, General, Bed shape Set button" is where you can change the bed graphic in Prusa Slicer, but Bambuslicer doesnt have this window anymore)
so the buildplate graphic would look like this

[image]

YifanWuBambu, 2022-11-17:
@ImagineerNL 
The calibration lines are associated with "start gcode", so changing start_gcode may cause the texture to be inconsistent with the actual print.

It would be a feature to parse "start gcode" and auto-render the calibration line lin in the preview page.

ImagineerNL, 2022-11-17:
I described it as 'optional' to select it.
Same as the 'avoid calibration room' when using auto-arrange, it could just be the 'default calibration room' to show.