bambulab/BambuStudio #645: A new instance opens every-time when exporting from Fusion 360 directly into BambuStudio (macOS)

onursagir, 2022-11-16:
When making a design that consists of multiple components/bodies in Fusion 360 you have the option to save the body/component individually, when saving you are offered the option to directly output to a 3D print utility. I have used this option plenty of times in the past using Cura however 1 little annoying thing has come up when using it with BambuStudio is that if I want to export 2-3 bodies individually the software opens up a new instance every time.

bambu123, 2022-11-21:
We think it's not a bug. Open with the following command line will start only a new Bambu Studio instance with all STLs from the command line loaded.

open -a BambuStudio ~/Downloads/snowflake-tie-fighter-kit-card-ornament-model_files/Individual\ Parts\ (Normal\ Tolerance)/*

onursagir, 2022-11-21:
However if you run the command twice it opens a new window instead of loading it into the active window. Usually in mac a new instance is not created unless the `-n` flag is provided from the man file:
```
     -n  Open a new instance of the application(s) even if one is already running.
```

bambu123, 2022-11-22:
OK, let's different 3mf and others such as STL files. If opening 3mf, always use a new instance (new window), others go to the current window.

onursagir, 2022-11-22:
@bambu123 yes how you describe it makes the most sense

YifanWuBambu, 2022-12-16:
This is included in v1.4.

closed 2022-12-16 by YifanWuBambu

onursagir, 2022-12-16:
Great work guys! Can't wait to try this