bambulab/BambuStudio #646: Filament complete unloads and loads each layer

AbeFM, 2022-11-16:
I've yet to fully characterize this, but printing has been nearly impossible for days now. It appears the printer is fully unloading and reloading each layer on a single material print.

On the particular print I'm doing now, I have "regular timelapse" turned on, though I suspect I was having these issues beforehand. I'm unsure if it is happening on every layer.

There is no reason to unload filament, why is this happening? Since the AMS often fails on load/unload (so far, only in these unnecessarily loads, I generally have good luck in MM prints), it means my 4 hour print has taken nearly a full day to come out because the printer is always errored out and there's always a tangle in my AMS.

Can I do something about this? Is the feeding/unfeeding on purpose?

YifanWuBambu, 2022-11-17:
Hi @AbeFM ,
If possible, could you share the 3mf file with us?

AbeFM, 2022-11-18:
Sadly, I cannot share one just now that I know does it for sure, I made some changes since my report. Is there a way to extract that from my print history?

My last few prints have worked ok, so until I can reproduce it with enough detail to track it down, we might we well leave this closed.

jhogendorn, 2023-10-30:
My printer has started doing this as well.

a .3mf file is here: https://www.dropbox.com/scl/fi/b3u3gtceyg5ohy6z5g3nf/gridfinity-hdd-bin.3mf?rlkey=ol1ntm4nvalhgo9j5wh2uaaml&dl=0

It's doing it regardless of project/particulr 3mf file etc.

Printer is X1C running 01.06.05.01, AMS is 00.00.06.40, Slicer is 1.7.7.89

I'm having a lot of issues with the ams failing to unload, and this bug makes it 10x more annoying because it keeps halting mid print for no reason.

ckempke, 2023-12-12:
Seeing the same thing.  Trying to print a large, single-material PETG model, and it's unloaded/reloaded the filament at least three dozen times so far, and failed ("AMS Motor is overloaded") on maybe a third of those.    Just hitting "continue" causes it to print again for a while, but it's making the printer almost unusable (at least for this print).

ckempke, 2023-12-12:
[Tablet_Stand_Apple-s.3mf.zip](https://github.com/bambulab/BambuStudio/files/13643281/Tablet_Stand_Apple-s.3mf.zip)

Adding a zipped 3mf file for my project.  (Github's comment system doesn't accept raw 3mf).