bambulab/BambuStudio #647: Bed "crashing" during start of next print after tall print puts bed at bottom of travel

JPotter73, 2022-11-16:
I printed a few tall pieces that ended with the bed at the bottom of the travel. After clearing and re-setting the build surface, I'd start the next print and noticed a low clicking as the bed tried to go lower and couldn't. As a work around, I raise the bed manually 20-40mm before starting the next print.

I'm not sure if it would be possible to know the bed is too low at the start of the next print to prevent this or automate an upward move

JPotter73, 2022-11-25:
Thought about it some more:
The dialog box should be displayed only after the printer did a long print (even if power cycled). It should simply ask if the bed is still at the bottom and if the user wants to raise it by 50mm or so.

This should put the power at the user's hand only when it's needed and to resolve it with a single click

JPotter73, 2022-12-14:
Additional Thought on this: use the camera (if equiped) to see if the build plate is in view (and clear?). If it isn't, raise the bed 10mm and check again. Repeat until in view at known height and then start homing.

dpinvidic, 2023-04-12:
I am  having this problem (firmware 01.03.00.00).  The firmware should simply check if the bed homing switch is active, and if not, try to rais the bed rather than lower it during initialization.