bambulab/BambuStudio #648: Add option to continue to next spool on filament runout

AbeFM, 2022-11-16:
When you get to the end of a roll of filament, would it be possible to have the AMS automatically load the next roll of compatible (same material) filament?

Example: Printing from AMS slot #2.
PLA PETg TPU PETg

If you run out of filament in #2, the AMS would notice you have PETg loaded in slot 4 and would seamlessly load and continue the print. This could result in objects with color bands, but having had a similar feature on the Prusa + MMU, I can tell you it is awesome for just finishing off spools.

YifanWuBambu, 2022-11-17:
@AbeFM 
This is in the plan.

Llorean, 2022-11-20:
Yeah, came here to request this same feature. At the very least, give an option to "connect" slots in the AMSes to mark them as valid substitutions for each other. So that way you can customize (like - use slot 3 when slot 2 runs out, but not slot 4, even though they're all marked as PLA). Especially useful since right now the AMS only lets us mark a very narrow amount of materials/brands, rather than labeling them custom materials. So PLA, PLA+, PLA with metallic particles or wood, etc, are all treated as the same basic material by the AMS.

Sonicboom247, 2024-01-17:
also came here to request the same thing, on the A1 i was not notified of a filament runout in the app, the printer was just sitting there with the bed on for about an hour or two..not only that but why have an app that has notifications that don't notify you of a filament runout, or a tangle? or anything other than print finished or canceled.