bambulab/BambuStudio #651: [Feature Request] Ability to upload sliced files to a queue without printing.

BambuLabDiscord, 2022-11-17:
Since you can't upload while the printer is printing, it's often handy for us who print all the time to prepare files ahead of time to make our flow easier. Could add a tab/page in the app similar to history for queue'd non-printed items, or simply add them to history with a flag of some sorts to note its not been printed yet.

YifanWuBambu, 2022-11-17:
We will first add a "send" button to queue sliced files in the next release.

sindrejl, 2022-12-05:
I would like to chime in on this request:

As a user of Raise3D printers I find the RaiseCloud Queue system extremely intuitive. What I really like is that on the printer, after I have cleaned out the finished print - on the screen I can just click "next print" and the printer will download next print from the cloud and start it up. It gives a very fluid workflow for people with either more then one machine or with many printjobs in queue

YifanWuBambu, 2022-12-16:
This is included in v1.4.

closed 2022-12-16 by YifanWuBambu

TomTomTomGit, 2023-12-09:
That's a great feature, thank you.

Is there a way to start a print on an P1P/P1S remotely?
And if you would implement such e feature, folder management would be great.