bambulab/BambuStudio #652: Please re-add shortcuts from PrusaSlicer

rkarlsba, 2022-11-17:
Hi all

In prusaslicer, you have shortcuts like <tab> for slice, + for add model etc. These seem to have been removed from BambuStudio. Can you please re-add them?

roy

YifanWuBambu, 2022-11-17:
Here are the bambu shortcuts.
[image]

If possible, please provide the missing frequently using shotcuts and we will add them.

rkarlsba, 2022-11-17:
Why not just add the lot from PrusaSlicer? https://help.prusa3d.com/article/keyboard-shortcuts_1764

YifanWuBambu, 2022-11-18:
Okay

rkarlsba, 2022-12-16:
I just upgraded to 1.4.0.17, but no new shortcuts there. Should it take so much time just to re-add these from the original prusaslicer source?

DomPerignon3, 2024-03-04:
Hello,

please add commands to switch between tabs, as discussed here https://github.com/bambulab/BambuStudio/issues/1515

Ex.:
Cmd-7 to 0 would be great and easy to learn.
Cmd-opt-number and/or Cmd-opt-left (or right) arrows to cycle through tabs, as web browsers.

Thanks.

nagy135, 2024-05-03:
i would also like to switch tabs with keyboard binds, it would let me not use mouse to just check how print is going ...macos binds with command+numbers would be the best but not sure what that means for windows users

JonLaliberte, 2025-01-23:
@YifanWuBambu Could we get this added. Switching between Prepare and Preview while making changes is annoying right now as it depends on the mouse.