bambulab/BambuStudio #653: cmake error

lotusrobotics, 2022-11-17:
experts@Alienware:~/work/projects/BambuStudio/deps/build$ cmake ../ -DDESTDIR="/home/experts/work/projects/BambuStudio_dep" -DCMAKE_BUILD_TYPE=Release -DDEP_WX_GTK3=1 make -j4
CMake Error: The source directory "/home/experts/work/projects/BambuStudio/deps/build/-j4" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.

lanewei120, 2022-11-17:
> experts@Alienware:~/work/projects/BambuStudio/deps/build$ cmake ../ -DDESTDIR="/home/experts/work/projects/BambuStudio_dep" -DCMAKE_BUILD_TYPE=Release -DDEP_WX_GTK3=1 make -j4 CMake Error: The [...]

these are two commands instead of one
[image]

I am sorry previous page don't have a clear ‘Line break'

lotusrobotics, 2022-11-17:
the "make -j4" has failed

experts@Alienware:~/work/projects/BambuStudio/deps/build$ make -j4
[  4%] Built target dep_Boost
[  8%] Built target dep_Qhull
[ 12%] Built target dep_FREETYPE
[ 13%] Performing configure step for 'dep_GLFW'
Scanning dependencies of target dep_PNG
Scanning dependencies of target dep_EXPAT
Scanning dependencies of target dep_JPEG
[ 13%] Creating directories for 'dep_PNG'
[ 14%] Creating directories for 'dep_EXPAT'
[ 14%] Creating directories for 'dep_JPEG'
-- Using Wayland for window creation
CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find WaylandProtocols: Found unsuitable version "", but required
  is at least "1.15" (found //usr/share/wayland-protocols)
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:376 (_FPHSA_FAILURE_MESSAGE)
  CMake/modules/FindWaylandProtocols.cmake:14 (find_package_handle_standard_args)
  CMakeLists.txt:257 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/experts/work/projects/BambuStudio/deps/build/dep_GLFW-prefix/src/dep_GLFW-build/CMakeFiles/CMakeOutput.log".
See also "/home/experts/work/projects/BambuStudio/deps/build/dep_GLFW-prefix/src/dep_GLFW-build/CMakeFiles/CMakeError.log".
CMakeFiles/dep_GLFW.dir/build.make:108: recipe for target 'dep_GLFW-prefix/src/dep_GLFW-stamp/dep_GLFW-configure' failed
make[2]: *** [dep_GLFW-prefix/src/dep_GLFW-stamp/dep_GLFW-configure] Error 1
CMakeFiles/Makefile2:852: recipe for target 'CMakeFiles/dep_GLFW.dir/all' failed
make[1]: *** [CMakeFiles/dep_GLFW.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 14%] No download step for 'dep_EXPAT'
[ 15%] Performing download step (download, verify and extract) for 'dep_JPEG'
[ 16%] Performing download step (download, verify and extract) for 'dep_PNG'
-- Downloading...
   dst='/home/experts/work/projects/BambuStudio/deps/build/JPEG/2.0.6.zip'
   timeout='none'
-- Using src='https://github.com/libjpeg-turbo/libjpeg-turbo/archive/refs/tags/2.0.6.zip'
-- Downloading...
   dst='/home/experts/work/projects/BambuStudio/deps/build/PNG/v1.6.35.zip'
   timeout='none'
-- Using src='https://github.com/glennrp/libpng/archive/refs/tags/v1.6.35.zip'
[ 17%] No update step for 'dep_EXPAT'
[ 17%] No patch step for 'dep_EXPAT'
[ 18%] Performing configure step for 'dep_EXPAT'
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    CMAKE_DEBUG_POSTFIX

-- Build files have been written to: /home/experts/work/projects/BambuStudio/deps/build/dep_EXPAT-prefix/src/dep_EXPAT-build
[ 18%] Performing build step for 'dep_EXPAT'
make[3]: warning: -jN forced in submake: disabling jobserver mode.
-- verifying file...
       file='/home/experts/work/projects/BambuStudio/deps/build/PNG/v1.6.35.zip'
[... 422 lines cut ...]
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/bin/wrjpgcom
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/share/doc/libjpeg-turbo/README.ijg
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/share/doc/libjpeg-turbo/README.md
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/share/doc/libjpeg-turbo/example.txt
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/share/doc/libjpeg-turbo/tjexample.c
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/share/doc/libjpeg-turbo/libjpeg.txt
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/share/doc/libjpeg-turbo/structure.txt
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/share/doc/libjpeg-turbo/usage.txt
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/share/doc/libjpeg-turbo/wizard.txt
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/share/doc/libjpeg-turbo/LICENSE.md
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/share/man/man1/cjpeg.1
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/share/man/man1/djpeg.1
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/share/man/man1/jpegtran.1
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/share/man/man1/rdjpgcom.1
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/share/man/man1/wrjpgcom.1
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/lib/pkgconfig/libjpeg.pc
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/lib/pkgconfig/libturbojpeg.pc
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/include/jconfig.h
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/include/jerror.h
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/include/jmorecfg.h
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/include/jpeglib.h
[ 26%] Completed 'dep_JPEG'
[ 26%] Built target dep_JPEG
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

lotusrobotics, 2022-11-18:
now tried with "make -j16" and got the following error

[ 89%] Performing dep_CGAL_version_fix step for 'dep_CGAL'
[ 90%] Performing dep_CGAL_relocation_fix step for 'dep_CGAL'
[ 90%] Completed 'dep_CGAL'
[ 92%] Built target dep_CGAL
HEAD is now at 9c0a8be1dc Merge branch 'gtk-initial-size'
[ 93%] Performing patch step for 'dep_wxWidgets'
/home/experts/work/projects/BambuStudio/deps/wxWidgets/0001-wxWidget-fix.patch:28: space before tab in indent.
    	, const wxWindow *window = nullptr
/home/experts/work/projects/BambuStudio/deps/wxWidgets/0001-wxWidget-fix.patch:30: space before tab in indent.
    	);
/home/experts/work/projects/BambuStudio/deps/wxWidgets/0001-wxWidget-fix.patch:41: space before tab in indent.
    	, const wxWindow *window = nullptr
/home/experts/work/projects/BambuStudio/deps/wxWidgets/0001-wxWidget-fix.patch:43: space before tab in indent.
    	)
/home/experts/work/projects/BambuStudio/deps/wxWidgets/0001-wxWidget-fix.patch:167: trailing whitespace.
        (anchorSideVertical & wxUP) || 
Checking patch build/cmake/init.cmake...
error: while searching for:
        else()
            find_package(OpenGL)
            if(WXGTK3 AND OpenGL_EGL_FOUND AND wxUSE_GLCANVAS_EGL)
                set(OPENGL_LIBRARIES OpenGL::OpenGL OpenGL::EGL)
                find_package(WAYLANDEGL)
                if(WAYLANDEGL_FOUND AND wxHAVE_GDK_WAYLAND)
                    list(APPEND OPENGL_LIBRARIES ${WAYLANDEGL_LIBRARIES})

error: patch failed: build/cmake/init.cmake:413
error: build/cmake/init.cmake: patch does not apply
Checking patch include/wx/fontutil.h...
error: while searching for:
    wxFontEncoding GetEncoding() const;

    void SetPointSize(int pointsize);
    void SetFractionalPointSize(double pointsize);
    void SetPixelSize(const wxSize& pixelSize);
    void SetStyle(wxFontStyle style);
    void SetNumericWeight(int weight);

error: patch failed: include/wx/fontutil.h:294
error: include/wx/fontutil.h: patch does not apply
Checking patch include/wx/gdicmn.h...
error: while searching for:
class WXDLLIMPEXP_FWD_BASE wxString;
class WXDLLIMPEXP_FWD_CORE wxIconBundle;
class WXDLLIMPEXP_FWD_CORE wxPoint;

// ---------------------------------------------------------------------------
// constants

error: patch failed: include/wx/gdicmn.h:38
error: include/wx/gdicmn.h: patch does not apply
Checking patch include/wx/generic/grid.h...
error: while searching for:
                         wxGridWindow* gridWindow);

    // Update the width/height of the column/row being drag-resized.
    void DoGridDragResize(const wxPoint& position,
                          const wxGridOperations& oper,
                          wxGridWindow* gridWindow);

    // process different clicks on grid cells
    void DoGridCellLeftDown(wxMouseEvent& event,

error: patch failed: include/wx/generic/grid.h:2951
error: include/wx/generic/grid.h: patch does not apply
Checking patch include/wx/msw/font.h...
error: while searching for:
    // ctors and such
[... 199 lines cut ...]
}

//
// contextual menus
//

error: patch failed: src/osx/cocoa/dataview.mm:1734
error: src/osx/cocoa/dataview.mm: patch does not apply
Checking patch src/osx/cocoa/settings.mm...
error: while searching for:
// ----------------------------------------------------------------------------

// Get a system metric, e.g. scrollbar size
int wxSystemSettingsNative::GetMetric(wxSystemMetric index, const wxWindow* WXUNUSED(win))
{
    int value;

error: patch failed: src/osx/cocoa/settings.mm:224
error: src/osx/cocoa/settings.mm: patch does not apply
CMakeFiles/dep_wxWidgets.dir/build.make:122: recipe for target 'dep_wxWidgets-prefix/src/dep_wxWidgets-stamp/dep_wxWidgets-patch' failed
make[2]: *** [dep_wxWidgets-prefix/src/dep_wxWidgets-stamp/dep_wxWidgets-patch] Error 1
CMakeFiles/Makefile2:154: recipe for target 'CMakeFiles/dep_wxWidgets.dir/all' failed
make[1]: *** [CMakeFiles/dep_wxWidgets.dir/all] Error 2
Makefile:90: recipe for target 'all' failed
make: *** [all] Error 2

lanewei120, 2022-11-18:
have you installed the components by following commands before
"sudo apt-get install cmake git g++ build-essential libgl1-mesa-dev m4 libwayland-dev libxkbcommon-dev wayland-protocols extra-cmake-modules pkgconf libglu1-mesa-dev libcairo2-dev libgtk-3-dev libsoup2.4-dev libwebkit2gtk-4.0-dev libgstreamer1.0-dev libgstreamer-plugins-good1.0-dev libgstreamer-plugins-base1.0-dev libosmesa6-dev"?

also when firstly built failed, we need to clean the patches of wxWidgets before start again:
cd dep_wxWidgets-prefix/src/dep_wxWidgets
git reset --hard
cd -
make -jN

lotusrobotics, 2022-11-18:
still same after "git reset --hard"

--------------------------------------------------------------------
 build/cmake/init.cmake...
error: while searching for:
        else()
            find_package(OpenGL)
            if(WXGTK3 AND OpenGL_EGL_FOUND AND wxUSE_GLCANVAS_EGL)
                set(OPENGL_LIBRARIES OpenGL::OpenGL OpenGL::EGL)
                find_package(WAYLANDEGL)
                if(WAYLANDEGL_FOUND AND wxHAVE_GDK_WAYLAND)
                    list(APPEND OPENGL_LIBRARIES ${WAYLANDEGL_LIBRARIES})

error: patch failed: build/cmake/init.cmake:413
error: build/cmake/init.cmake: patch does not apply
Checking patch include/wx/fontutil.h...
error: while searching for:
    wxFontEncoding GetEncoding() const;

    void SetPointSize(int pointsize);
    void SetFractionalPointSize(double pointsize);
    void SetPixelSize(const wxSize& pixelSize);
    void SetStyle(wxFontStyle style);
    void SetNumericWeight(int weight);

error: patch failed: include/wx/fontutil.h:294
error: include/wx/fontutil.h: patch does not apply
Checking patch include/wx/gdicmn.h...
error: while searching for:
class WXDLLIMPEXP_FWD_BASE wxString;
class WXDLLIMPEXP_FWD_CORE wxIconBundle;
class WXDLLIMPEXP_FWD_CORE wxPoint;

// ---------------------------------------------------------------------------
// constants

error: patch failed: include/wx/gdicmn.h:38
error: include/wx/gdicmn.h: patch does not apply
Checking patch include/wx/generic/grid.h...
error: while searching for:
                         wxGridWindow* gridWindow);

    // Update the width/height of the column/row being drag-resized.
    void DoGridDragResize(const wxPoint& position,
                          const wxGridOperations& oper,
                          wxGridWindow* gridWindow);

    // process different clicks on grid cells
    void DoGridCellLeftDown(wxMouseEvent& event,

error: patch failed: include/wx/generic/grid.h:2951
error: include/wx/generic/grid.h: patch does not apply
Checking patch include/wx/msw/font.h...
error: while searching for:
    // ctors and such
    wxFont() { }

    wxFont(const wxFontInfo& info);

    wxFont(int size,
           wxFontFamily family,

error: patch failed: include/wx/msw/font.h:23
error: include/wx/msw/font.h: patch does not apply
Checking patch include/wx/msw/tooltip.h...
error: while searching for:
    // the one and only one tooltip control we use - never access it directly
    // but use GetToolTipCtrl() which will create it when needed
    static WXHWND ms_hwndTT;

[... 184 lines cut ...]
}

//
// contextual menus
//

error: patch failed: src/osx/cocoa/dataview.mm:1734
error: src/osx/cocoa/dataview.mm: patch does not apply
Checking patch src/osx/cocoa/settings.mm...
error: while searching for:
// ----------------------------------------------------------------------------

// Get a system metric, e.g. scrollbar size
int wxSystemSettingsNative::GetMetric(wxSystemMetric index, const wxWindow* WXUNUSED(win))
{
    int value;

error: patch failed: src/osx/cocoa/settings.mm:224
error: src/osx/cocoa/settings.mm: patch does not apply
CMakeFiles/dep_wxWidgets.dir/build.make:122: recipe for target 'dep_wxWidgets-prefix/src/dep_wxWidgets-stamp/dep_wxWidgets-patch' failed
make[2]: *** [dep_wxWidgets-prefix/src/dep_wxWidgets-stamp/dep_wxWidgets-patch] Error 1
CMakeFiles/Makefile2:154: recipe for target 'CMakeFiles/dep_wxWidgets.dir/all' failed
make[1]: *** [CMakeFiles/dep_wxWidgets.dir/all] Error 2
Makefile:90: recipe for target 'all' failed
make: *** [all] Error 2

lanewei120, 2022-11-18:
the above error is just "patch failed in wxwidgets"
have you gone to the correct directory and then "git reset --hard"

after you execute this command
please execute "git status" and list the result here for a check

lotusrobotics, 2022-11-18:
experts@Alienware:~/work/projects/BambuStudio/deps/build$ cd dep_wxWidgets-prefix/
experts@Alienware:~/work/projects/BambuStudio/deps/build/dep_wxWidgets-prefix$ git reset --hard
HEAD is now at 96707fc FIX: tool change time is too long
experts@Alienware:~/work/projects/BambuStudio/deps/build/dep_wxWidgets-prefix$ git status
On branch master
nothing to commit, working tree clean

lanewei120, 2022-11-18:
the directory is wrong from your above information

it should be /work/projects/BambuStudio/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets

lotusrobotics, 2022-11-18:
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/lib/libwx_baseu-3.1.a
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/lib/libwx_baseu_net-3.1.a
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/lib/libwx_gtk3u_core-3.1.a
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/lib/libwx_gtk3u_adv-3.1.a
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/lib/libwx_gtk3u_aui-3.1.a
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/lib/libwx_gtk3u_html-3.1.a
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/lib/libwx_gtk3u_propgrid-3.1.a
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/lib/libwx_gtk3u_ribbon-3.1.a
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/lib/libwx_gtk3u_richtext-3.1.a
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/lib/libwx_gtk3u_webview-3.1.a
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/lib/wx/3.1.5/web-extensions/webkit2_extu-3.1.so
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/lib/libwx_gtk3u_xrc-3.1.a
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/lib/libwx_gtk3u_media-3.1.a
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/lib/libwx_gtk3u_gl-3.1.a
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/lib/libwx_gtk3u_qa-3.1.a
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/lib/libwx_baseu_xml-3.1.a
-- Installing: /home/experts/work/projects/BambuStudio_dep/usr/local/bin/wxrc
[ 96%] Completed 'dep_wxWidgets'
[ 96%] Built target dep_wxWidgets
Makefile:90: recipe for target 'all' failed
make: *** [all] Error 2

lanewei120, 2022-11-18:
wxWidgets built successfully

need more logs to check which part has error

closed 2023-01-11 by lanewei120