bambulab/BambuStudio #655: [Feature Request] Smooth timelapse without purgeblock

TheCableGuy99, 2022-11-17:
I'm not sure if there's something I'm missing but why does it need to build a tower/purge block when doing a smooth timelapse?

Why can it not just move to a position, take a pic and continue printing? It does this in Octolapse (An octoprint plugin for timelapses).

Can this be considered? 

Thanks.

YifanWuBambu, 2022-11-23:
It will be allowed in the next release. However, the melted filament may drop from the nozzle and be left on the outer surface of the model if there is no wipe tower.

TheCableGuy99, 2022-11-23:
I see now why it's needed, maybe wipe it on the filament bin wiper? Not sure if that would work? p.s. I think the bin wiper needs just a little work. Quite often you'll find filament doesn't always end up in the bin. Not a huge issue but if it can be worked on to make it more reliable then great :)

YifanWuBambu, 2022-11-23:
The bin wiper may not work well enough.
If some filament drop from nozzle, that means the nozzle may have some empty area inside. This may cause lack of extrusion at the beginning of model after each photo for the timelapse.

TheCableGuy99, 2022-11-23:
Fair enough, you guys are the experts so i'll leave it to whatever you decide is best. I just couldn't understand why a purge tower was being used but I get it now.

YifanWuBambu, 2022-11-24:
You can watch the details of the purge tower. There are some defects on the apperance, caused by the above problems that I describes. If no wipe tower, these defects may go to the model.

TheCableGuy99, 2022-11-24:
I think we should close this as a bad idea then.

YifanWuBambu, 2022-11-26:
Glad to discuss with you : )

closed 2022-11-26 by YifanWuBambu

YifanWuBambu, 2022-12-16:
This is included in v1.4.