bambulab/BambuStudio #656: [stop one print on plate instead of entire print.] stop one failing print not entire print

BambuLabDiscord, 2022-11-17:
i would like to be able to stop one object on the bed if it fails instead of entire print. will help save filament.

YifanWuBambu, 2022-11-23:
This is in our plan list, but may be not in recent releases.

jacycatlin, 2022-12-15:
It's the one Octoprint feature I miss the most often, by far.

SaltWei, 2023-08-02:
This feature has been included in latest V01.06.00.00 X1 series firmware and [v01.07.01.62 studio](https://github.com/bambulab/BambuStudio/releases/tag/v01.07.01.62). And please make sure both firmware and studio is updated.

Currently it only work in one filament per plate situation, and doesn't support multi filament because of the complexity of flushing and changing filament. One filament per plate situation can still benefit a lot when has some objects failed to be printed and need to be skipped.

Thank you everyone for this good suggestion!

closed 2023-08-02 by SaltWei