bambulab/BambuStudio #658: [Feature Request] Upload sliced model to Folder or SD Card via Bambu Studio

neogenix, 2022-11-18:
Hi, 

It'd be wonderful to be able to slice and upload to a destination (folder, or SD Card) without printing. This is similar to #651 which is mostly around a queue system (which I would also love fwiw). I would love to be able to slice things and upload them to my printer as I'm going along instead of attempting to upload an individual, or hauling the SD card around. 

Bonus points if you could slice all build plates in a project, and upload them all to a folder / project?

Merlwynd, 2022-11-18:
+1 ...at least for the first part of the above request...

terryrobison, 2022-11-20:
On my tool changer printer, I slice and upload all parts to be printed to the printer.   Then I can print each component from the printer or network.  With the Bambu, I have to slice each part, print, slice then print.  Slicing them and copying the files to a TINY card is not a good solution.  Why didn't they make this a USB if they wanted you to slice and carry jobs to the printer?

YifanWuBambu, 2022-11-23:
@neogenix 
This will be included in the next release.

YifanWuBambu, 2022-12-16:
This is included in v1.4.

closed 2022-12-16 by YifanWuBambu

Merlwynd, 2022-12-16:
When trying to load a print to printer SD card, I keep getting the message that the printer must be part of the same LAN network as the slicer. The printer and slicer are part of my home wifi network....why do I get this message?

YifanWuBambu, 2022-12-16:
@Merlwynd 
Like stated in the release note, please use this feature with printer upated to v1.3 (comming soon).

Buujakka99, 2022-12-20:
I have version 01.03.00.00 on my printer. When I want to send a file to the SD card, the message always comes up: Send to Printer failed. How can I fix the problem?