bambulab/BambuStudio #660: [Feature Request] Set printer defaults that persist between uses

neogenix, 2022-11-18:
Hi! 

If I'm printing the same object over and over, I tend to deselect some of the default options. It would be good for us to be able to set the printer to automatically remember these. Perhaps a settings window where these can be persisted on the cloud with a config the printer pulls periodically.

YifanWuBambu, 2022-11-23:
Hi @neogenix
Thanks for the suggestion. For firmware issues, please move to https://bambulab.com/en/support for better support.

neogenix, 2022-11-26:
@YifanWuBambu before I do so, I was thinking about more when you send a print from studio it doesn’t persist options there.

YifanWuBambu, 2022-11-26:
@neogenix 
Okay. We may provide an option in preference to determine wether to remember last used option values.