bambulab/BambuStudio #661: [Feature Request] Improve color options in Filament Setup

neogenix, 2022-11-18:
Hi!

Today, we have a limited number of colors, it'd be nice to either give us a few options like a color wheel, or just increase the color options (and include metalic, etc...)

aridShrub, 2022-11-18:
I agree ; metallic, translucent filaments need more than just a colour patch.

I would like to see the Bambu Lab colour names, e.g. “PLA Matte Ivory White” ; vs. “PLA Basic White” both display “PLA” and a white colour bar, so you cannot differentiate between the two. Maybe if you hover over it , additional detail could be given.

There is a Bambu Lab transparent” filament” - PC Black - and it is listed as “Black” which is confusing.

I’d like to be able to enter an RGB for third party filament. I use the Apple colour names now -  pretty good but not “peachy”

[image]

aridShrub, 2022-11-19:
Bambu Lab PC "Black" filament is actually transparent.