bambulab/BambuStudio #662: [FIRMWARE/SLICER] Excessive vibration affecting quality, can limit it via slicer but not remove it completely

H4kunaM4tata, 2022-11-18:
The models are showing these weird marks not present on the STL file.
I have moved the printer to a more solid surface, did the calibration but it didn't fix it
This line on the left side does not exist on the stl file.

[image]

The while test shows extreme ghosting, not present with the black colour.

[image]

I can reduce the ghosting/ridging/whatever it is called by reducing the printer speed.
As you can see, I have reduced speed overall by 50% or more.
The default profile has worse quality.
I have performed the test with more users to isolate my printer as the problem, their printers show the same deep line on the left side of the M letter.
Something seems off.

[image]

H4kunaM4tata, 2022-11-19:
Further test, as you can see the pic, reducing the speed improves it but there's still a hardware creating the deep line.

SaltWei, 2022-11-25:
1. The belts need to be re-tensioned and the Vibration Compensation needs to be calibrated again. [Please follow this guide](https://wiki.bambulab.com/x1/maintenance/belt-tension) to do it. 
2. Lower the acceleration value to 5000mm/s², under Speed → Acceleration → Normal Printing
3.  Loosen the 3 screws of the extruder and tighten them again. After tightening the screws, manually shake the hot end to see if it is loose.

4. Raise the heating bed to a higher position, but be careful not to hit the nozzle. Loosen the six screws that hold the heating bed in place as shown, then tighten them again.
https://wiki.bambulab.com/x1/maintenance/belt-tension

By the way, this is not the right place to report printer issue. You can turn to APP to report issue.

closed 2022-11-25 by SaltWei