bambulab/BambuStudio #663: [SLICER] Slicer engine doesn't seem to be processing models properly

H4kunaM4tata, 2022-11-18:
This has been by far the worst example.
The printer only prints "okay" high quality models.
Models I am used to just download, slice on PrusaSlicer and print, now are showing some terrible skin artifacts.
Or even Cura for the matter, seems to be slicing better .

BambuLab doesn't seem to be using Arachne engine either.
You have to hack sometimes increasing the layer width so the slicer can replace internal solid infill with solid infill for example.

Models I design and save with high quality is fine, but the ones I download and would normally just print, are having terrible quality.
BL Slicer should process STL files just as good as PrusaSlicer or Cura, not worse.

5h print using 0.20 profile.

[image]

SaltWei, 2022-11-23:
> BL Slicer should process STL files just as good as PrusaSlicer or Cura, not worse.

Do you mean the stl is sliced by Prusa and cura and then printed with the same Bambu Printer?

H4kunaM4tata, 2022-11-23:
> > BL Slicer should process STL files just as good as PrusaSlicer or Cura, not worse.
> 
> Do you mean the stl is sliced by Prusa and cura and then printed with the same Bambu Printer?

Hey @SaltWei , that means models with low quality mesh are printed better, better processed with those slicer and their respective printers than BambuSlicer/X1.
It is fair to say that I never had to worry about how good quality was a model mesh with SlicerPrusa/MK3S+, just slice and print.
X1C tho requires that level of management.
It makes no much sense to use another slicer other than the printer's slicer for the matter.

At the moment, I am down by 50% speed overall and other users have reported improvement using my profile.

I am guessing this problem won't get sorted out any time soon, as longer as you guys know the problem is there is a fix, you should be okay to close it.

H4kunaM4tata, 2023-01-10:
This ringing is far worse than I expected.
Find a SLA or high quality poly only, X1C cannot print low poly models, no matter how good other printer might print the very same model.
I am closing this

closed 2023-01-10 by H4kunaM4tata