bambulab/BambuStudio #664: [FIRMWARE] Stopped print job does not stop the back fan

H4kunaM4tata, 2022-11-18:
If I cancel a printer for whatever reason, the printer does not stop the back fan, it will keep running at full blast.
If you are printing PLA which turns it on, the fan will never stop, unless you turn off the printer.

For ASA, you don't want that fan at full speed,
In theory, the gcode instruction should overwrite but that is theory.

This need to be fixed.

c2h5oh, 2022-11-18:
Even a filament retraction failure at the end of completed print is enough to keep the fan running.

YifanWuBambu, 2022-11-23:
Hi @H4kunaM4tata ,
Thanks for the suggestion. For firmware issues, please move to https://forum.bambulab.com/ for better support.

H4kunaM4tata, 2022-11-23:
@YifanWuBambu , here is the best place to track issues, that forum is new, barely has any useful post.
This is not a simple post but a firmware issue, no better place than GitHub.
Thanks

maxmax3, 2022-11-30:
Needs to be fixed. It's annoying.
Thanks

closed 2023-01-10 by H4kunaM4tata