bambulab/BambuStudio #665: part cooling fan not turning on after set layer

gRIMMIE1981, 2022-11-18:
my part cooling fan is not turning on after the set layer.  is there a bug of some sort when using the ams?  I set the fan in the generic pla for what i am using

SaltWei, 2022-11-23:
Please provide 3MF file for analyzing

gRIMMIE1981, 2022-11-23:
I'm all set with this thank you

Tom J

SaltWei, 2022-11-25:
Not understand.
Already has no issue now?

gRIMMIE1981, 2022-11-25:
I figured the part cooling fan out. but my extruder is dead and I am waiting to see what and when they are shipping parts

Tom J

SaltWei, 2022-11-25:
I have checked with my printer and everything is OK. So this should not be issue about Studio.
So close.
Thank you for reporting issue and suggestions for us to keep improvement!

SaltWei, 2022-11-25:
By the way, for issue about printer, you can report on App. 
https://wiki.bambulab.com/en/x1/troubleshooting/upload-log

closed 2022-11-25 by SaltWei