bambulab/BambuStudio #666: [Feature request] Add a Calibrated All Slots Operational Mode

DSW4546Admin, 2022-11-18:
Currently, the calibration only perform the calibration of the 1st filament that is used.  This is fine if all filaments are by the same make/model/batch and has mininal differences between them.  However, often in using mix color/filament mode, 1 color will run out first.  So unless the replacement filament is in the 1st slot, the new filament will not be calibrated and will depends on the previous filament's previous calibration values.  So please add an operational mode either via the Printer Control (UI) or via the Bambu Studio app to perform either a 'selectable' calibration or 'All Slots' calibrations to pick up all 4 slots calibration values.

Thanks.

Merlwynd, 2022-11-18:
I understand what you are saying, but not sure how BambuLab would practically make that work. Where on the plate would the other calibrations be done?…or would you want to ‘pause’ between filaments to remove the calibration lines on the plate?….seems a bit self defeating of the automatic AMS function..

DSW4546Admin, 2022-11-18:
That's why I said a separated Calibrated All Slots Operational Mode.  If I have left that part out, sorry for the confusion.  There are more than enough room to do 4 of those calibration print on the full plate if it's an individual/separated operational.

Merlwynd, 2022-11-18:
You had that in…yes, there is enough calibration room to do all those calibrations on the plate at the same time, but you would still have to remove them before the print actually starts unless it is a very small print?

DSW4546Admin, 2022-11-18:
Maybe I'm still not being cleared.  A SEPARATED UI action initiated on the Printer, say under the AMS, like where currently there's the Load and Unload individual filament.  Not asking for any changes to how file are sent over and can do the calibration on the initial filament.  So that if you were to load in a NEW filament, you won't have to print something with that filament to get its calibration values but you just click on this on the printer UI and it will do all 4 slots (for those that might think this is too much waste of other filament, fine, add a selection box of 1 to 4 and an 'All').  There's no Load + Calibrate for each filament that you load (I suggest everyone to hit up the load to test that the roll is place in the AMS correctly).

DzzD, 2023-01-14:
+1, also retaining/saving calibration of each individual filaments in the BambuStudio would be great to be able to re-use it later by attaching this calibration to a filament preset.