bambulab/BambuStudio #667: [Feature Request] Exclude print region

BambuLabDiscord, 2022-11-19:
On my previous printer in octoprint, i had a plug in to exclude region on the build plate (link: https://plugins.octoprint.org/plugins/excluderegion/ ) mid pring if someting went wrong with one of the objects you could exclude that region there the object was, which was quite useful not to waste filament on half printed objects that were fine.

YifanWuBambu, 2022-11-23:
Canceling one or more objects during printing is in our plan. But may be not in recent releases.

RidoKilos, 2023-01-05:
Yes! Love to see this acknowledged by the team. This will be an amazing addition to the software! 💪

SaltWei, 2023-08-02:
This feature has been included in latest V01.06.00.00 X1 series firmware and [v01.07.01.62 studio](https://github.com/bambulab/BambuStudio/releases/tag/v01.07.01.62). And please make sure both firmware and studio is updated.

Currently it only work in one filament per plate situation, and doesn't support multi filament because of the complexity of flushing and changing filament. One filament per plate situation can still benefit a lot when has some objects failed to be printed and need to be skipped.

Thank you every one for this good suggestion!

closed 2023-08-02 by SaltWei