bambulab/BambuStudio #669: [Feature Request] add filament selection for brim and calibration (or default calibration to most used filament)

tsdexter, 2022-11-19:
I would like to be able to explicitly select which filament to use for the brim. For example, in this print it decided to use the white for the brim even though all of the outside edges are in red... This means if any bits of the brim are not coming off very well it is very noticeable because it's white. If the brim was red, this wouldn't be an issue. 

Along the same idea, I would also like to select which filament to calibrate. Generally, I use the same types of filaments just different colors so which is calibrated doesn't necessarily matter however, that's not always the case and I would like to be able to select the red for calibration since the model is 99% red. At the very least, it should know to calibrate whichever filament is used the most. For example this model is 14 grams of red and <1g of white so it'd have the best result if the red is properly calibrated. 

[image]

tsdexter, 2023-09-28:
looks like the new (?) color sequence feature will solve this problem. Not sure when that was added and why it wasn't mentioned here. 

Can you confirm that brim is always printed with color sequence #1? If so, then changing the sequence solves the problem. 

[image]

SaltWei, 2023-10-08:
The calibration lines and purge line at the edge of plate is always using the first used non-support filament.
So changing the first layer filament sequence will influence which filament will be calibrated.

tsdexter, 2023-10-08:
completed via filament sequence option addition

closed 2023-10-08 by tsdexter

SaltWei, 2023-10-08:
Thank you very much for suggestions!